micropython/micropython #1330: esp8266: pyb.RTC differences with pyboard implementation

owens-bill, 2015-06-15:
I've found differences between the esp8266 and pyboard implementations of pyb.RTC that make them incompatible:

 - The pyboard code implements pyb.RTC as a class definition with a constructor, for esp8266 it is an instance of the class (I hope I'm describing that distinction correctly):
```
Micro Python v1.4.3-159-g3ce212e-dirty on 2015-06-14; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import pyb
>>> pyb.RTC
<class 'RTC'>
>>> rtc = pyb.RTC()
>>> rtc
<RTC>
```
```
Micro Python v1.4.3-133-g7ed58cb-dirty on 2015-06-10; ESP module with ESP8266
Type "help()" for more information.
>>> import pyb
>>> pyb.RTC
<RTC>
>>> rtc = pyb.RTC()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: object not callable
```

 - The pyboard 8-tuple includes the week day as 1-7 for Monday-Sunday, esp8266 uses 0-6.

 - The last entry in the pyboard 8-tuple is a 'SubSeconds' counter that goes down from 255 to 0 during each second. The last entry in the esp8266 8-tuple is a microseconds counter that goes up from 0 to 999.

danicampora, 2015-06-15:
> The last entry in the pyboard 8-tuple is a 'SubSeconds' counter that goes down from 255 to 0 during each second. The last entry in the esp8266 8-tuple is a microseconds counter that goes up from 0 to 999.

You mean milliseconds, right? I did the same for the CC3200 since I think it's more useful to return milliseconds than subseconds

owens-bill, 2015-06-15:
> You mean milliseconds, right? I did the same for the CC3200 since I think it's more useful to return milliseconds than subseconds

Sorry, yes - milliseconds. And I don't yet have a CC3200 board running (a LAUNCHXL is sitting on my desk waiting for me to read the docs on how to flash it) so I didn't add that to the comparison, but looking at the code I see that you're also in the 0-6 camp for weekdays.

danicampora, 2015-06-15:
> but looking at the code I see that you're also in the 0-6 camp for weekdays.

That's correct. I thought it would made more sense to work the same way as `time.localtime()` which output weekday in the 0-6 range.

owens-bill, 2015-06-15:
There's an interesting difference in how esp8266 and pyboard handle the initial setting of the RTC that affects time.localtime() as well. If I give pyb.RTC a naive tuple to start it up, setting just the date but leaving the time and the week day zeroed out, it does the right thing:
```
>>> rtc.datetime((2014,1,1,0,0,0,0,0))
>>> rtc.datetime()
(2014, 1, 1, 2, 0, 0, 1, 828)
>>> time.localtime()
(2014, 1, 1, 0, 0, 4, 2, 1)
```
1/1/2014 was a Wednesday, so week day == 2 for the esp8266. However, if I do the same for the pyboard, it keeps the incorrect week day, and that throws off time.localtime()
```
>>> rtc.datetime((2014,1,1,0,0,0,0,0))
>>> rtc.datetime()
(2014, 1, 1, 0, 0, 0, 1, 11)
>>> time.localtime()
(2014, 1, 1, 0, 0, 3, -1, 1)
>>> rtc.datetime((2014,1,1,3,0,0,0,0))
>>> rtc.datetime()
(2014, 1, 1, 3, 0, 0, 1, 84)
>>> time.localtime()
(2014, 1, 1, 0, 0, 3, 2, 1)
```
The second time I gave it the correct week day number (3 in this case) and all is well. 

There's nothing in the docs to indicate whether any of the fields are optional, so I think it's reasonable to assume that they should all be correctly specified when setting the RTC, but it is still a bit nicer if the code can fix things up when needed.

dhylands, 2015-06-15:
mktime is the C runtime library function used for setting time. The documentation for mktime http://linux.die.net/man/3/mktime says: "The function ignores the values supplied by the caller in the tm_wday and tm_yday fields."

It also says that it will correct the values for tm_wday and tm_yday. It also normalizes things, so that you can do stuff like use October 40 and it will be converted to November 9. This is allow you take adate like October 30, add 10 days to it and mktime will normalize appropriately.

MicroPython's mktime behaves the same way.

So, I think that it would be most useful if the RTC time setting routines took the same arguments as mktime.

owens-bill, 2015-06-16:
Regarding the difference between milliseconds counting up and "subseconds" down from 255, I wonder whether either one is worth exposing in this context. We have pyb.millis() to get the uptime in milliseconds if you want to time something and the RTC drifts so quickly, on the esp8266 at least, that specifying the time to the nearest second is questionable, much less going to milliseconds. And of course the esp8266 RTC is only good for seven hours regardless.

dpgeorge, 2015-06-17:
The reason pyboard RTC().datetime takes 1-7 for the weekday and 0-255 for the subseconds is because that's how the hardware behaves and this function is a simple getter/setter of the hardware values.  It's more like the `hwclock` command under Linux than the `date` command.

The proper way for the user to access the date/time is via time.localtime() [and time.mktime()].  These are standardised, and as @dhylands says above do correction of input values.  They don't expose milliseconds in the RTC, but, as @owens-bill says, use pyb.millis() for that.

Some remarks:
1. ESP8266's RTC constructor should be modified to work the same as stmhal's; ie `rtc = pybRTC()`
2. stmhal, cc3200 and esp8266 all provide time.localtime() which conforms to the Python standard.  Use it for accessing the date/time.
3. rtc.datetime() should be used to get/set the hardware RTC values, and will in general be port specific.  That said, we can try to make it as similar as sensible across ports; for example I can change whe weekday to be in the range 0-6, but this will break existing code.

dpgeorge, 2015-06-22:
pyb.RTC() now constructs a (singleton) RTC object on esp8266.

pfalcon, 2016-05-28:
This should be fixed in revamped port.

closed 2016-05-28 by pfalcon