micropython/micropython #1331: Suggested easy TODOs for contributors

pfalcon, 2015-06-16:
List of low-hanging fruit people interested in contributing can start with:

* We need minimal base64 encode/decode implementation.

mvdw73, 2015-07-09:
What about dma for the sd card on pyboard (stm)? Is anyone working on this?

blmorris, 2015-07-09:
DMA for the sd card would be good to have, IMHO. I was going to attempt it after wrapping up my I2S work, but it's just as well if you get to it first.
I took a quick hack at it a couple months ago without success; since then the DMA support code has been refactored so it should be easier to use.

tkschmidt, 2015-10-04:
I would like to try to contribute to this project. 
At coveralls.io I just see a percentage of test coverage. Is there also a way to see what is not covered?

wgreybe, 2016-09-21:
Hi. I'm not at the level where I can add value, as Im just starting to learn to code, but Id like to be part of this in order to hopefully learn (assuming I can get my head around programming).

pfalcon, 2018-02-16:
ucollections.deque is implemented in my fork: https://github.com/pfalcon/micropython/commits/pfalcon and serves as a base of uasyncio 2.0: https://forum.micropython.org/viewtopic.php?f=15&t=85&start=80#p25186

reidwagner, 2018-02-17:
Okay. Glad I asked!

cr0mbly, 2020-03-22:
> Refactor to use more efficient API calls to decrease code size (example: fe3cc5b)

I'm keen to do this (Python developer but wanting to learn C and this project excites me), is this just a straight find and replace of the above `nlr_raise` calls too `mp_raise_ValueError` where where a ValueError is required.

Thanks in advance!

iggy12345, 2020-09-07:
Is anybody working on making import statements more flexible? Would it be a good idea to add relative import statements?

peterhinch, 2020-09-07:
Perhaps you could explain what type of relative imports where you are having problems. In my experience they work. Taken from working MicroPython code:
```python
from . import local
from .check_mid import CheckMid
```

dpgeorge, 2022-03-30:
Closing because this issue is outdated.

closed 2022-03-30 by dpgeorge