micropython/micropython #1337: Elaborating github project title

pfalcon, 2015-06-18:
Following the latest conventions for naming, it would be nice to change name in github description to MicroPython. Also, would be nice to make description more useful for search results, etc. Suggested: "MicroPython, a small Python implementation for microcontrollers and constrained systems".

dpgeorge, 2015-06-18:
Yes we can remove the space between u and Py; see eg 3eece29807f5257091271229dbf41543646eb4e4 for effort in this direction.

I don't think the word "small" conveys the meaning we want it to, I'd rather "compact" or "lean" or "resource conscious".

pfalcon, 2015-06-18:
Thanks, just to clarify, I meant github project description as seen e.g. at https://github.com/micropython/micropython close to the top of page (currently reads "The Micro Python project").

dpgeorge, 2015-06-18:
Yes, I understand.  There are lots of places to change it.

dpgeorge, 2015-06-23:
All github titles now updated (incl organisation and micropython-lib).

closed 2015-06-23 by pfalcon