micropython/micropython #1343: Location in docs for inline-assembler reference

dpgeorge, 2015-06-22:
@peterhinch has recently written a nice reference for uPy's inline assembler, and it's ready to merge into the main docs.  I think it belongs in the main docs because it documents a part of the code in the py/ directory (emitinlinethumb.c).  This reference is valid for all ports that are a Thumb2 architecture and that have the inline assembler enabled.  As such it shouldn't go in the pyboard/ specific directory of the docs.  It's also not suitable for the library/ directory.  Following CPython I suggest a directory reference/ (within the docs/ directory).

Any other ideas?

pfalcon, 2015-06-22:
I agree it belongs to the main docs (well, ideally, would split to generic "assembler functions" reference and arch-specific assembler syntax, but not now).

dpgeorge, 2015-06-24:
Inline assembler reference added in 2110dc5a6dc9de415e8f4dda75262d4b98d1f0a4.

closed 2015-06-24 by dpgeorge