micropython/micropython #1347: Incorrect exception type raised when MICROPY_ERROR_REPORTING_TERSE is selected

danicampora, 2015-06-23:
Hi,

While running the tests on the WiPy I have seen that some of the basic ones fail. I don't know if the best is to raise tickets or submit PRs or both, so please advice. Here's the first one: `int1.py`

At some point the test expects a `ValueError` raised, but `SyntaxError` is thrown instead. Because in `parsenum.c` line 144:

```c
value_error:
    // if lex!=NULL then the parser called us and we need to make a SyntaxError with traceback
    if (MICROPY_ERROR_REPORTING == MICROPY_ERROR_REPORTING_TERSE) {
        mp_obj_t exc = mp_obj_new_exception_msg(&mp_type_SyntaxError,
            "invalid syntax for integer");
        raise_exc(exc, lex);
    } else {
        mp_obj_t exc = mp_obj_new_exception_msg_varg(&mp_type_ValueError,
            "invalid syntax for integer with base %d: '%.*s'", base, top - str_val_start, str_val_start);
        raise_exc(exc, lex);
    }
```

I guess that in both cases it should be `mp_type_SyntaxError`, at the test needs to be corrected, right?

danicampora, 2015-06-23:
CPython returns `ValueError`

dpgeorge, 2015-06-23:
Yes it should be a ValueError, feel free to change it :)  With this change I find that all basic tests pass with ERROR_REPORTING_TERSE (using unix port).

danicampora, 2015-06-23:
Cool, thanks :-)

danicampora, 2015-06-23:
Done.

closed 2015-06-23 by danicampora