micropython/micropython #1349: esp8266: Where, how, or whether to document esp_init_data_default.bin

owens-bill, 2015-06-23:
Briefly mentioned in the comments for PR# 1267 was the existence of a file called esp_init_data_default.bin, which is the only way to change the behavior of several of the underlying ESP8266 SDK functions that MicroPython depends on. The discussion in 1267 was around reading the ADC pin, which can either measure an external level (0-1 volt) or the supply voltage, with the choice being determined - and calibrated - based on one byte in esp_init_data_default.bin. Other bytes in the file are used for two functions not yet exposed: system_deep_sleep_set_option() and system_phy_set_max_tpw(). I'm guessing that there are still other uses for the file, not yet documented.

I knew that this was a thing because I've read the commit comments and the SDK docs, but they are not at all clear about esp_init_data_default.bin, and it took a fair amount of Googling to find the source of the file (the default version is in the SDK zip file) and how to write it to the board (esptool.py can do it). 

I think that the MicroPython documentation for pyb.ADC for the ESP8266 should at least mention that you need to choose between measuring Vdd or the external pin. Should there be more information about the file and how to change the selection? Should it also describe how to upload the new file? Should we have a tool that generates esp_init_data_default.bin with the appropriate value? Or should people just be expected to search for this information?

Note that this isn't a MicroPython problem exclusively; the Arduino folks are having some difficulty figuring out how to make it integrate cleanly into their environment: https://github.com/esp8266/Arduino/issues/338  They seem to have come down on the side of choosing the default (external ADC) and putting it into their kit that way, with no further documentation.

Testato, 2015-06-24:
hi, i'm not using micropython, but i'm very interested on ADC problem on ESP8266, and i try to give my support to igrr (author of arduino core porting)
The official documentation is clear on this point, you can use Tout pin for analogRead OR read the voltage of power supply. You cannot do the two function toghether because the Tout ADC (pin17) must be Floating for measure the vddread.

Until now many error is present on this part, because the official documentatio was not clear, but now we know that the way for choosing whitch function we want from Tout pin is done by the 107th byte of the init bin file.
So the first way can be create two init bin file and upload the correct one depending from which of the two function thge user need.
The second way, choosed by @igrr is create an array of the value that the ESP need and upload it instead of uploading the init bin file.
This is a very good idea, because we do not need upload a separated bin file, and it is not important what it is flashed from factory to the init bin file memory section.

the work is at beginning, it is here  https://github.com/esp8266/Arduino/commit/1803f3052afed566c2698110aa5b447e628bc924#diff-3776add87f5c7f96ac02b7fbfef5305aR204

Testato, 2015-06-24:
This is the official explanation of the 107th byte that set the use of Tout ADC Pin:

FOR ANALOGREAD:
system_adc_read
Function:
Measure the input voltage of TOUT pin 6, unit：1/1024 V
system_adc_read is only available when wire TOUT pin to external
circuitry，Input Voltage Range restricted to 0 ~ 1.0V.
• The 107th byte in esp_init_data_default.bin(0〜～127byte) is named as
“vdd33_const”, and when wire TOUT pin to external circuitry, the
vdd33_const must be set as real power voltage of VDD3P3 pin 3 and 4.
• The range of operating voltage of ESP8266 is 1.8V~3.6V，the unit of
vdd33_const is 0.1V，so effective value range of vdd33_const is [18,
36].

FOR POWER SUPPLY READ:
system_get_vdd33
Function:
Measure the power voltage of VDD3P3 pin 3 and 4, unit：1/1024 V
Note:
• system_get_vdd33 can only be called when TOUT pin is suspended
• The 107th byte in esp_init_data_default.bin（0〜～127byte）is named as
“vdd33_const“ , when TOUT pin is suspended vdd33_const must be set as
0xFF， that is 255

Testato, 2015-06-24:
I added this explanation to the 107th byte:

// vdd33_const
    // the value of this byte depend from the TOUT pin usage (1 or 2):
    // 1) 
    // analogRead function (system_adc_read()):
    // is only available when wire TOUT pin17 to external circuitry, Input Voltage Range restricted to 0 ~ 1.0V.
    // For this function the vdd33_const must be set as real power voltage of VDD3P3 pin 3 and 4
    // The range of operating voltage of ESP8266 is 1.8V~3.6V，the unit of vdd33_const is 0.1V，so effective value range of vdd33_const is [18,36]
    // 2) 
    // getVcc function (system_get_vdd33):
    // is only available when TOUT pin17 is suspended (floating), this function measure the power voltage of VDD3P3 pin 3 and 4
    // For this function the vdd33_const must be set as 255.
    [107] = 33,

igrr, 2015-06-25:
@owens-bill I have pushed a change which allows you to select ADC mode at a sketch level by adding either `ADC_MODE(ADC_TOUT);` (which is the default) or `ADC_MODE(ADC_VDD);` line in the global scope. I'm not sure though how this may be applied to micropython, as my implementation relies on some features of gcc and ld, namely weak functions and function wrapping.

closed 2015-10-25 by pfalcon

flukeee, 2017-08-04:
Hi there - I have an application where the hardware MUX's several analog inputs to the ADC pin.  The MUX can float its output.  In this setup, we satisfy the condition to leave the TOUT pin floating.  I would like to change the configuration byte 107 during runtime, even performing a system reset if I have to, in order to be able to read the ADC voltage.
Thanks.

peterhinch, 2017-08-05:
This is the bug tracker for MicroPython. If you're using MicroPython queries like this should be raised in the [forum](https://forum.micropython.org/). To measure the supply voltage I'd suggest applying it (with a suitable voltage divider) to a MUX input.