micropython/micropython #1354: unix: list all builtin modules/features

grimnight, 2015-06-28:
Currently there is not a way to list all builtin/frozen modules.
You need to read through the source to find them.
What if you need to know if the current micropython binary has a module you need?

Python REPL has `help("modules")` which lists all available modules, which would be useful for micropythons REPL, it could also support `builtin_modules` and `frozen_modules` values.

Using `vim --version` outputs all the compiled in features in vim, micropython could have that too.
Why not have a flag that lists all builtin modules and/or frozen modules?.

pfalcon, 2015-06-28:
> Why not have a flag that lists all builtin modules and/or frozen modules?.

Simple question, simple answer: because it's bloat. But we'll have sys.builtin_module_names sooner or later (as soon as someone contributes it).

pfalcon, 2015-06-28:
> What if you need to know if the current micropython binary has a module you need?

Use Python:

~~~~
try:
    import module_I_need
except ImportError:
   print("Oops, there's no module I need")
~~~~

grimnight, 2015-06-28:
Since micropython can have frozen modules and extmods, how can you distinguish one micropython binary from another? How can you tell what has been compiled in? Or are you just going to guess what modules there are available?

You could add a wiki entry listing available modules by compile option.
You could also add a list of available modules by micropython version, but that would only be useful if there was a version entry for micropython in `sys`.

pfalcon, 2015-06-28:
> Since micropython can have frozen modules and extmods, how can you distinguish one micropython binary from another?

You pose wrong questions. It's not "how", it's "why". Feel free to argue why it's needed and if you prove that the needed is unavoidable, it's most certainly will be accepted when someone contributes implementation.

Please also make sure you're familiar with "zen" of MicroPython: https://github.com/micropython/micropython/wiki/ContributorGuidelines

ARF1, 2017-02-24:
@pfalcon This functionality exists in the code now. However for the esp8266 port it is not enabled. (MICROPY_PY_BUILTINS_HELP_MODULES does not appear in `mpconfigport.h` to overwrite the no-set default from `mpconfig.h`)

Is this deliberate or did this fall through the cracks when the MICROPY_PY_BUILTINS_HELP_MODULES was introduced?

Would it be ok, to enable it for the esp8266 by default? It would definitely help novice users.

dpgeorge, 2017-02-24:
> Is this deliberate or did this fall through the cracks when the MICROPY_PY_BUILTINS_HELP_MODULES was introduced?

When new features are added to the core they aren't automatically added to all the ports.

But since help('modules') is so useful, I'd be +1 on adding it to esp8266.

pfalcon, 2017-02-24:
+1 from me too.

dpgeorge, 2017-02-24:
Ok, help('modules') was enabled for esp8266 in 6771adc75fc795c909fb3371b376906ef4285919

It's (still) not available on unix port because unix doesn't even have the help builtin (although it's now easy to add that just by turning on some other config options).

ARF1, 2017-02-24:
In PR #2906 I enabled `help('modules')` for the unix and windows builds. (With the exception of the unix nan-boxing verision which fails to build with the help in general.)

pfalcon, 2017-02-24:
Unix port, just the same as before, can implement any help() or any other builtin with a Python module (subject to cleaning up all the boot.py vs site.py issues, if any remaining).

Jongy, 2019-02-05:
See https://github.com/micropython/micropython/pull/4468

dpgeorge, 2021-07-13:
`help("modules")` was enabled in 6430cd3e02d4ec4a8fc949754b434f745b1973e6 for the unix port.

I'll close this issue on the back of that commit, but it's still interesting to consider #4468.

closed 2021-07-13 by dpgeorge