micropython/micropython #1355: unix: disable or limit module importing

grimnight, 2015-06-28:
Would it be possible to get commandline flags to disable or limit module importing?
A commandline switch that disables all or parts of importing and another switch that permits certain modules (builtin/frozen).
Such flags could be used from scripts or shebang lines `#!`.

pfalcon, 2015-07-03:
MicroPython implements Python language and takes CPython as a reference. Does CPython have such commandline flags? In either case, this feature would extensive discussion. Otherwise, direct answer to the question would be: "Maybe, but why?"

pfalcon, 2015-07-03:
Taking command-line flags out of the question, it's a generic sandboxing question, #1356.

closed 2015-07-31 by pfalcon