micropython/micropython #1356: Sandboxing

grimnight, 2015-06-28:
One of the great benefits of Lua is sandboxing, is there a plan to support sandboxing in some way?
Would it be possible to provide a limited/custom environment for `eval`?

pfalcon, 2015-06-28:
> One of the great benefits of Lua is sandboxing

Is that really the reason why people use/choose Lua? Do you have references to that, and to specific projects which use that feature?

> is there a plan to support sandboxing in some way?

So far nobody needed it.

> Would it be possible to provide a limited/custom environment for eval?

Python generally has good enough builtin sandboxing model. All needed features may not be yet implemented in uPy, but given a good usecase, patches would be accepted. Alternatively, any builtin function can be compiled out (we're not yet there, but that's the aim), including eval().

pfalcon, 2015-06-28:
Also, https://github.com/micropython/micropython/issues/1355 appears to be related to sandboxing either. Can you please leave only one of these open?

dpgeorge, 2015-06-28:
To what extent would you consider it to be sandboxed?  Assuming you're talking about the unix port, the only way code can "escape" outside the uPy process is by using the ffi module (and in that case it has access to pretty much anything that it can do as the executing user), or os.system, or the socket module.  
You can easily compile uPy without these modules.

EcmaXp, 2015-06-28:
Check https://github.com/EcmaXp/micropython
You can control CPU Time and Memory (already by gc with heap, but my port can use multi state in one process)
And custom import function (by enable override bulitin module) also supported.

Just copy unix port and modify for sandboxing.

EcmaXp, 2015-06-28:
Example is opencom port, but problem is there is no document.

pfalcon, 2015-09-13:
Scope is not clear, no response from submitter, closing for now.

closed 2015-09-13 by pfalcon