micropython/micropython #1360: TAB does not work in the interactive interpreter

xix-xeaon, 2015-07-02:
I know there's already an issue about this, but it was closed without resolving the problem. Tab completion is now enabled by default in Python 3, and it was a mistake. It was supposed to include code allowing tab indentation at the beginning of lines since tab completion for everything is pointless anyway. It is currently being [fixed](https://bugs.python.org/issue23441), but, tabbing in the micropython interactive interpreter does not yield any completion either. Of course, that's a perfect feature to strip for a smaller footprint, but then there's no reason for tab not to work for indentation.

As such, if there's no tab completion anyway, it should be off by default. Although that will eventually be resolved with the aforementioned fix. However, since there's already no tab completion, but also no tab indentation that means tab does not work as it should, in either case.

*inb4 "spaces is the only way!": PEP 8 is a style-guide, not a law. Blindly following it is "the Hobgoblin of Little Minds". The fact of the matter is that the entire point of the tab character is indentation. If you actually think space works better for you, then that's great, go ahead. If you're only using it because of PEP 8, well... I use tab because it's better for indentation than spaces, and it also allows anyone to view the code with the indentation size they prefer. The reason it's in PEP 8 is very much related to the maximum line length, and they're both completely irrelevant since we don't use 80 character displays anymore.*

pfalcon, 2015-07-02:
There's tab completion in uPy, it's just @dpgeorge chose to not do completion against "builtins" module. This waits for extended criticism and patch to add configurable completion for it too.

There was also feature request for smart auto-completion vs indentation. Note that once it's implemented, it will almost certainly make "Tab" produce standard 4-space indent. If you need real tabs, use "raw" input mode.

stinos, 2015-07-02:
> I use tab because it's better for indentation than spaces

Please let's not go over that discussion again. Anyway I don't understand why you include this here: this is about the repl. What does it matter if indentation *on the repl* is done using spaces or tabs??

xix-xeaon, 2015-07-02:
@pfalcon, if that's the case then it would be great. Any timeline on this?

@stinos, unfortunately every time someone brings up an issue with tabs in any python community the response is almost always "PEP 8! Use SPACES!" as if tabs aren't even supposed to work.

pfalcon, 2015-07-02:
> Any timeline on this?

You know about it as much as I do - when somebody who really needs it, up to implementing it, will implement it, it will be there.

> "PEP 8! Use SPACES!"

That's because programming is social activity. If some people will pronounce "a" as "o", while other as "u", they will hard time to communicate. Just the same here, using spaces is the best compromise to get everyone to *see* the same code and be able to cooperate. Anything else can be seen as effort to make it harder for other people to see/use your code, and surely, people suggest against that.

pfalcon, 2015-07-02:
>> Any timeline on this?

Or do you mean raw mode? It's supported on stmhal, but doesn't seem to be on unix, that needs to be fixes, and we generally accept and document it not as "internal, machine communication" feature, but as a general-purpose one, for humans too (it's useful for cop-pasting code to REPL for example).

xix-xeaon, 2015-07-02:
@pfalcon, yeah copying code is a big issue. I don't get raw mode. I can enter it but nothing happens. I get that it doesn't echo, but none of the code I paste seem to run either. Printing, naming labels, toggling leds. Nothing.

dpgeorge, 2015-07-02:
> I can enter it but nothing happens.

You need to send an EOF char (ctrl-D) when you've finished entering code.  Newline sends a literal newline and doesn't finish input.

xix-xeaon, 2015-07-02:
@dpgeorge, ah, yes I see. Well that'll work until tab works in normal mode, thanks!

BTW, I'm very happy with the device, and not just because it runs Python! Compared with Arduino this device seems better designed. Simply the fact that you upload your program through a standard USB storage device was one of the things that got me to order one. Arduino is actually rather cumbersome to work with, but with micropython everything is really easy and quick. Also, all the ports for various communication protocols are great. I hope it becomes more popular so more libraries for driving different external components become available.

dpgeorge, 2015-07-02:
@xix-xeaon don't forget to checkout the forum for drivers: http://forum.micropython.org/viewforum.php?f=14

dpgeorge, 2015-07-06:
Ok, I have a patch with the following behaviour:
- on a line starting with ">>>" tab always does auto completion
- on a line starting with "..." tab will "indent" if all characters before the cursor are spaces (includes the case if the cursor is at the start of the line), otherwise it will auto complete
- "indent" means: insert up to 4 spaces to move the cursor to a position which is the next multiple of 4 (ie standard tabbing behaviour)

Comments on this?

xix-xeaon, 2015-07-07:
Yeah that's great! +1 all the way.

Some details though: I assume indent will always insert at least 1 space? (That it, it will never just move "through" the spaces.) And the multiple counts the spaces after the cursor as well? Does (should?) it move the cursor to the end of all the spaces, or just to the end of the inserted (or completed, in case of 1-3 spaces) indent?

Also, if you could include shift-tab to do the opposite the feature would really be complete, although it's more of a convenience thing when you accidentally a tab and don't want to press backspace 4 times (because it's so much work ;) ).

I don't think smart backspace/delete nor smart arrow keys have much of a place here, I mean it's not an editor. You mostly just want to be able to paste any code and write some simple things.

pfalcon, 2015-07-07:
@dpgeorge : This was discussed in #1200, I now submitted #1368, and wonder if we can shoot straight for it.

dpgeorge, 2015-10-11:
Auto indent is implement along with paste mode.  Physical tabs work in paste mode.  Is that enough to close this issue?

pfalcon, 2015-10-25:
> Auto indent is implement along with paste mode. Physical tabs work in paste mode. Is that enough to close this issue?

I hope it is, closing.

closed 2015-10-25 by pfalcon