micropython/micropython #1363: esp8266: network functions only work when REPL is active

deshipu, 2015-07-03:
Due to the event-driven nature of the ESP code, the network-related operations (such as sending to and receiving from sockets) only happen when the REPL console is waiting for input, and now when user code is being executed. That means, that it's not currently possible to have a program which would, for instance, periodically check a pin and send a network message when it changes state -- we would need to use a loop with a delay, and that means the REPL never gets a chance to run, so the message is never actually sent. To send it, the program would need to exit to REPL, but that means, that it couldn't start working again without human intervention. Since there are no timers, it's impossible to do it in any other way.

I'm not familiar with the SDK enough to suggest a working solution, but it would be great if calling delay() also let the network functions run. Alternatively, calling some special function, that would pump the events, if that's possible.

dpgeorge, 2016-01-30:
This will be fixed when #1806 is closed.

pfalcon, 2016-03-24:
esp8266 switched to lwIP sockets.

closed 2016-03-24 by pfalcon