micropython/micropython #1364: Unable to login to the wiki

dhylands, 2015-07-03:
This isn't a micropython issue, but I wasn't sure where else to post it.

I was trying to edit the page: http://wiki.micropython.org/official-accessories/Bluetooth-module-HC05
and when I click on login, I get a page where I can choose from github/googleoath2 and OpenID.

Clicking on github results in a 404 error.
Clicking on GoogleOAuth2 results in a 400 error: Missing required parameter: client_id with the following details:

    scope=https://www.googleapis.com/auth/userinfo.email
    response_type=code
    redirect_uri=http://wiki.micropython.org/__omnigollum__/auth/google_oauth2/callback
    access_type=offline
    state=f8e9ac4f978f9293230cab731b319af4f05f844efffd9f4f
    client_id=

dpgeorge, 2015-07-05:
It's also reported on the forum: http://forum.micropython.org/viewtopic.php?f=2&t=780

I'll get to it as soon as I can.

dpgeorge, 2015-07-29:
Fixed.

closed 2015-07-29 by dpgeorge