micropython/micropython #1365: Default build broken on systems without Python 3

roger-, 2015-07-05:
The official requirements say:

> You will also need bash and Python (at least 2.7 or 3.3).

However ../tools/make-frozen.py requires Python 3 (see [here](http://buildbot.openwrt.org:8010/broken_packages/ar71xx/micropython/compile.txt), for example).

pfalcon, 2015-07-05:
Thanks for the reports and proper references. Please provide suggestions how to fix it - is recent python2.7 available there, so `from __future__ import print_function` would work, and we just switch shebang to "python"?

roger-, 2015-07-05:
Doesn't look like there's a build option to disable frozen modules? Maybe there should be.

Anyway, I think 2.7 is a reasonable assumption so that would probably work.

pfalcon, 2015-07-05:
Of course there's. By unix version of micropython now comes with builtin package manager, period. (If it's not there, it's not a micropython.)

pfalcon, 2015-07-06:
Fixed in 3a2e9f20f60561f9b43d0abc4ec38ea62a0b3ef5. Please let know if that fixes openwrt build.

closed 2015-07-06 by pfalcon

dpgeorge, 2015-07-06:
The "#!" still needs to be changed, but otherwise I confirm it works with Python2.

pfalcon, 2015-07-06:
Yep, sorry, there were issues in 1st commit, and shebang was changed back and forth locally... Now everything should be ok.