micropython/micropython #1368: Smart autoindent/autocompletion

pfalcon, 2015-07-07:
As a followup to #1200, #1360, I'd like to propose that following behavior was implemented for REPL (in a sense, if anything is implemented, then following behavior):

1. If pressing Enter at the end of normal line which starts with number of 4-spaces, then next line automatically has that number of spaces inserted (auto-indent).
2. If line ends with ":" (starts compound statement), next line auto-indent with increased amount of indent.
3. If line was just auto-indented, pressing backspace removes 4 spaces (can be repeated).

With the scheme above:

4. Tab can always do completion. If user mistakenly presses backspace in auto-indented, one is just "punished" with pressing space 4 times.

And:

5. With all the advanceties above, we'd now need "paste mode" to disable all these features and allow pasting existing code into repl. It essentially should be like "raw" mode, but with echo.

xix-xeaon, 2015-07-07:
I think autoindent is a good feature, but I don't think having to switch mode in order to paste code is a good thing. I also don't consider being able to tab to auto"complete" on whitespace-or-nothing preceding cursor to be a completion-feature. It's a "please list everything because I can't even remember the first letter of the thing I want"-feature, and I think being able to tab instead of quad-space is way more valuable than that.

I suppose you could detect if it was a manual or pasted newline on time since last input or something. Pasted code would newline immediately after the last character, while a manual newline would take "some time" (in comparison). Maybe complicated though =P

dpgeorge, 2015-07-07:
These are nice features that certainly help writeability.  But it adds quite some complexity in the implementation because now you need additional state (eg paste/no-paste, was line just auto indendent?, what was the last line?) and there is stronger coupling between the readline function and the REPL.

Note that currently readline is independent of how the REPL works: readline just reads one line, and the REPL asks for multiple lines if it detects something like a function definition or unclosed bracket.

Losing direct paste ability (without doing something special to enter paste mode) hurts: I know that lots of people use pasting as a way to enter simple code, and they'll need to relearn (after being frustrated with paste doing funky things).

How will paste mode work?  We could use ctrl-E to enter/exit it (or ctrl-E to enter, ctrl-F to exit so you know that ctrl-E always enters no matter what mode you're in).

xix-xeaon, 2015-07-07:
What about turning it around, and having an autoindent mode? That way normal mode would be like regular python, but if you're gonna be typing a lot then you could switch to autoindent mode at the cost of having to switch back if you want to paste blockcode. If autoindent in the future becomes smart enough to allow pasting of blockcode then it could be merged into normal mode.

pfalcon, 2015-07-07:
> Losing direct paste ability (without doing something special to enter paste mode) hurts

There's no paste ability already, try e.g. paste following:

~~~~
def foo():
    print(1)
def bar():
    print(2)
~~~~

So, we'd need "paste mode" anyway if we want to make it truly easy for people to paste, without relying whether there was truly empty line between functions, etc.

> How will paste mode work? We could use ctrl-E to enter/exit it

My initial idea was to just use raw mode, I missed that it doesn't echo. But otherwise yes, just having single key combo (+ visual feedback) should work, as it's human-intended feature, so no extra robustness with separate start/end keys is needed imho.

pfalcon, 2015-07-07:
> but I don't think having to switch mode in order to paste code is a good thing

Sorry, you can't have both - it's either comfortable to type in new code (all indentation done automagically for your), or paste existing code with existing indentation. Trying to "autodetect" what's being done now won't be reliable in general case.

pfalcon, 2015-07-07:
> What about turning it around, and having an autoindent mode?

Most people will type code, only few will paste large enough amounts of code, and those are to learn that special paste mode exists.

pfalcon, 2015-07-07:
> But it adds quite some complexity in the implementation because now you need additional state (eg paste/no-paste, was line just auto indendent?, what was the last line?) 

Well, those are just 1-bit flags.

> and there is stronger coupling between the readline function and the REPL.

Well, can't say much about this. I of course have TODO to play and try to implement it myself to see how it would go, but so far didn't get to it. But I hope you agree that the above gives pretty much IDE-like experience.

xix-xeaon, 2015-07-07:
> There's no paste ability already, try e.g. paste following:

That doesn't work in regular python either, but I think it should be rather easy to fix. And if you care about PEP 8 there really should be blank lines in between ;)

@pfalcon, I don't think a lot of people really work with a lot of blockcode (where you need indentation) in a one-line REPL. I'll write a lot one-liners in the REPL, but as soon as it becomes just a couple of lines in a block I'll work with the code in an editor and then I'll copy paste it into REPL and play (test) with it using one-liners.

If you wanna have a multiline REPL then it's a different thing, but I think we're outside of the scope of the implementation, for now anyway.

pfalcon, 2015-07-07:
> I don't think a lot of people really work with a lot of blockcode (where you need indentation)

Yeah, that's certainly true for "big OS" system, where it's few clicks/keystrokes to start editing a file, and it's easier to dump code there. But with pyboard and other bare-metal ports, there's no such opportunity. There're other approaches to that still, e.g. http://forum.micropython.org/viewtopic.php?f=15&t=669 , http://forum.micropython.org/viewtopic.php?f=15&t=708 , but in the end, builtin REPL is the only thing you're guaranteed to have at any particular moment. So, for me, the changes above are simple natural extension to existing REPL stuff (which already has auto-completion, detection of block statements, etc.) Maybe it's not that simple of course.

dpgeorge, 2015-07-07:
> There's no paste ability already, try e.g. paste following:
> 
> def foo():
>     print(1)
> def bar():
>     print(2)

You are supposed to enter one function at a time!  At that's what people do.

What you point out above is a different issue, not related to tabbing: the parse mode of the input needs to change from SINGLE_INPUT to FILE_INPUT for that example to work.  And this departs a lot from the standard way of the REPL working.

So is that what we want, a paste mode that is completely different, and is basically raw mode with echo?

pfalcon, 2015-07-07:
> You are supposed to enter one function at a time!

We're not talking about "entering", but about "pasting". It may be different situation, like there's blank line, but with few spaces in it, etc. etc.

> the parse mode of the input needs to change from SINGLE_INPUT to FILE_INPUT for that example to work.

Yes, that's how "paste" mode should work, if we decide to implement it.

> So is that what we want, a paste mode that is completely different, and is basically raw mode with echo?

Yes, I would find it useful. Consider the bare port, which lacks (e.g. so far), filesystem, networking, etc. - how to do some useful work with it? By letting people literally paste in (big enough) amounts of code.

dpgeorge, 2015-07-07:
> We're not talking about "entering", but about "pasting"

Yes, and pasting/entering one function at a time currently works, and is currently used by many people.

> Yes, that's how "paste" mode should work, if we decide to implement it.

We cannot break the current useful and working behaviour of pasting without providing an alternative.

dpgeorge, 2015-09-12:
Auto indent implemented; see #1434.  Still left to do is paste mode.

dpgeorge, 2015-10-11:
Paste mode implemented.  Auto-indent now enabled for stmhal port.

closed 2015-10-11 by dpgeorge

methoxid, 2016-01-30:
Before this change, pasting a code properly indented with spaces always worked in the old REPL mode, only tabs were problematic. 
Now it is not clearly visible in terminal which whitespaces are being used, WHY is there suddenly a tab character, or if it is 4 spaces instead. Moreover, there is inconsistency in whitespaces: if you write a function in the smart repl mode, it inserts Tab character, however if you copy the resulting code and paste into an editor, there are 4 spaces instead. 
I do not think one should re-invent iPython on stmhal, let's focus on important python core issues and keep it simple, stupid. The functionality of SMART indentation shall be kept to the user-space software such as terminal / editor / IDE, which will be required sooner or later. The simple approach allows what stmhal is good for: automation scripting in python. 
One usecase which demonstrates the strengths of stmhal is: send short python code over UART to repl and execute, this is not possible with smart repl, it would require the user to maintain two codebases like this: 
Old REPL, working fine, **function1.py** :
```
def a():
....return 1
```
New smart REPL **function1_smartrepl.py** which can be copy/pasted over UART without the added complication of ctrl-E and Ctrl-D, which in some cases can trigger soft reset:
```
def a():
return 1
```
Basically: this feature request just invented a new language. If users want the smart repl, they shall program it themselves in main.py. Calling python form command line does not smart indent. Calling iPython does. iPython is not in core python for a reason, it is just a userspace program.

danicampora, 2016-01-30:
I disagree with completely. Autocompletion and autoindent are great and work like a charm. If you need to paste code, just use the paste mode ;-) (which also works like sunny!)

peterhinch, 2016-01-30:
@danicampora +1 In daily use the REPL improvements are awesome.