micropython/micropython #1375: unix usocket: External representation of (binary) addresses

pfalcon, 2015-07-11:
Ok, https://github.com/micropython/micropython/pull/1002 was closed, because approach of requiring usage of getaddrinfo() to pass addresses to socket functions certainly works. But there's opposite problem - how to show to user an address received from a socket function, e.g. accept(), which doesn't have any solution now. I can imagine following choices:

1) Give up and return Python-standard tuple representation of addresses, then almost certainly also need to accept them to all other functions. Bunch more extra code, memory required for minimal socket server grows.
2) Stay stubborn and make sockaddr a separate object, which will do conversion only on str() or repr(). No extra RAM overhead with this solution, but object will be more or less involved, e.g. standard Python tuple should be emulated.
3) Stay stubborn and low-profile - just provide a function to convert binary address to Python tuple.

Thoughts?

dpgeorge, 2015-07-23:
I was just using: `remote_ip = "%u.%u.%u.%u" % (peername[4], peername[5], peername[6], peername[7])`.  It works but you need to know the raw format of the addr, and also that its IPV4 (which is indicated in the raw addr).  This would be option 4: don't add anything :)

See lib/netutils for functions that already convert IPV4 info to/from a tuple.

SpotlightKid, 2015-11-20:
I would suggest implementing a conversion function in usocket (i.e. 3) and use that in micropython-socket to wrap e.g. `recvfrom`.

pfalcon, 2015-11-20:
@SpotlightKid didn't mention that this, but this came from his thread on the forum: http://forum.micropython.org/viewtopic.php?f=15&t=1161 .

pfalcon, 2015-11-20:
>  It works but you need to know the raw format of the addr, and also that its IPV4 (which is indicated in the raw addr). This would be option 4: don't add anything :)

You can't know that it's IPv4 and can't even know format of the raw sockaddr - it depends on libc and architecture, and generally unpredictable what it may be on different systems.

pfalcon, 2015-11-20:
Ok, I'm torn between making a sockaddr type, and leaving it a bytes, but adding decoding function. I'm sure that if a function is selected, it later will need to be converted to type anyway. But it allows to save many bytes on type definition so far...

Another question is what it should decode to. My original idea was to use CPython's socket address representation, but the more I think about it, the more completely adhoc thinh it looks. E.g. AF_INET address is (host, port), while AF_NETLINK is (pid, groups). In other words, seeing a tuple of 2, you can't really say for  address family it is, without external context.

So, I leave towards following implementation: sockaddr() takes a raw sockaddr bytes, and return tuple of (af, net_addr[, port] [, anything else]). net_addr is still bytes, and need to be run thru inet_ntop() (to be implemented too) to produce ascii address.

The drawback of this is that it's a long way for a human to get human-readable address (not very pythonic). But it's still a giant leap from not being to do that to being able, and it allows to do that correctly for any address family. (A sceptic may of course ask if that makes sense - realistically, only AF_INET and AF_INET6 would be used thru generic socket interface, anything else would likely use custom wrappers. But well, that what makes me select an easy to compile-out function instead of whole type.)

pfalcon, 2015-11-20:
Btw, sokaddr being a type is how I implemented it initially and factored away later, there're still remnants of that in unix/modsocket.c.

pfalcon, 2015-11-20:
sockaddr() implemented in 3ba61656bdc725e6dda46f05b5a65247b3ce7925