micropython/micropython #1381: viper type error doesn't issue a line number

peterhinch, 2015-07-13:
The compile time error message  
```ViperTypeError: can't do binary op between 'object' and 'int'```  
doesn't report the line in error.

dpgeorge, 2015-07-23:
Yes, there is no capability at the moment to give line numbers.  That's because the emitter framework doesn't know anything about the parsenode (which stores the line number info) corresponding to the emit call.  For bytecode its not a problem because all errors are handled by the compiler which does know the parsenode.  But with a ViperTypeError, this is raised in the emitter, and to get line number info the parsenode would somehow need to be registered before the emit call.

In can be done but requires some thought.

dpgeorge, 2015-07-27:
ViperTypeError now includes the filename and the function name of the error.  The line number corresponds to the start of the function definition that had the error.

To give the exact line number of the location of the error would require more work.   One idea would be to hook into the compile function for a "suite" and check at each statement in the suite for a compile error, adding a traceback with the approximate line number if one was raised.

peterhinch, 2015-07-28:
Anything which enables the line to be identified is a big step forwards!

dpgeorge, 2015-07-29:
After thinking about it I managed to improve compiler line number error reporting for bytecode compilation which also improved the situation for viper error reporting.

There were some rare/obscure cases with bytecode compilation errors that gave no line number which are now fixed (or at least improved to give a sensible line).

ViperTypeError will now most of the time will report the correct line number.  At the very least the line number will be the line number of the start of the function that has the error.

closed 2015-07-29 by dpgeorge