micropython/micropython #1383: Viper code produces incorrect results

peterhinch, 2015-07-15:
The following produces correct results in Cpython, MicroPython and MicroPython with the native code emitter. With viper the code runs  but the result is all zeros. I have broken the function down into three smaller ones which do work under viper, hence the long code example. I suspect the example code runs too fast under viper: it executes considerably faster than running the three smaller ones in succession: overall runtime (dominated by this function) drops from 5.7S to 3.6S. Perhaps code is being optimised out of existence?
```python
BYTES_PER_LINE = const(10)
LINES_PER_DISPLAY = const(10)
BYTES_PER_SCAN = const(5)
class Bar():
    def __init__(self):
        self.image = bytearray(BYTES_PER_LINE*LINES_PER_DISPLAY)
        self.linebuf = bytearray(BYTES_PER_LINE * 2 + BYTES_PER_SCAN)

    @micropython.viper
    def foo(self, line: int, pixelmask: int, offset: int):
        index = 0
        buf = self.linebuf
        image = self.image
        for b in range(BYTES_PER_LINE, 0, -1):
            buf[index] = int(image[offset + b - 1])  ^ pixelmask | 0xaa
        index = BYTES_PER_LINE
        scan_pos = (LINES_PER_DISPLAY - line -1) >> 2
        scan_shift = (line & 3) << 1
        for b in range(BYTES_PER_SCAN):
            buf[index] = 3 << scan_shift if scan_pos == b else 0
            index += 1
        index = BYTES_PER_LINE + BYTES_PER_SCAN
        for b in range(BYTES_PER_LINE):
            pixels = ((int(image[offset + b])  ^ pixelmask) >> 1) | 0xaa
            pixels = (((pixels & 0xc0) >> 6)
                | ((pixels & 0x30) >> 2)
                | ((pixels & 0x0c) << 2)
                | ((pixels & 0x03) << 6))
            buf[index] = pixels
            index += 1

a = Bar()
a.foo(0,0,0)
print(a.linebuf)
```

peterhinch, 2015-07-16:
To add to this I've done the obvious test and measured the execution time of foo() above. With the native decorator it takes 372uS. With viper 31uS: presumably little of the code is being executed.

pfalcon, 2015-07-16:
@peterhinch: I'd consider  @micropython.viper usage in a class as "too advanced" and suggest to avoid it. Write plain functions with it, call them from you class (same as you'd do for a function coded in C). The usage like above may work some time, but somebody will need to submit a lot of patches first.

peterhinch, 2015-07-16:
@pfalcon  Thanks very much indeed for that, the function works as you suggested (in 256uS) if I instantiate the bytearrays as globals. The only reason the function was an instance method was to access the bytearray members. Short of putting these in global can you suggest a way for an method to pass member bytearrays to a plain viper function? I've had no success trying to pass them as function arguments.

pfalcon, 2015-07-16:
I don't know much about viper, just general idea is that it supports (and has the most performance effect) on simple constructs. Trying to produce simple example which work/don't work as expected would help. And do you know about ptr8/ptr16 annotations? - https://github.com/micropython/micropython/blob/master/tests/micropython/viper_ptr8_load.py

peterhinch, 2015-07-16:
Thanks for that - I hadn't spotted that. However it does seem that viper doesn't like having several function arguments. I now have a simple example of code which works until you add the viper decorator:
```python
@micropython.viper
def get1(src:ptr8, dest: ptr8, dat: int, dat2: int, dat3: int) -> int:
    dest[1] = src[1] + dat + dat2 + dat3
    return src[1]

c = bytearray(4)
b = bytearray([0,1,2,3,4])
print(get1(b, c, 20, 1, 1))
print(c)
```
which produces
```
>>> import rats3
0
bytearray(b'\x00\x00\x00\x00')
```

pfalcon, 2015-07-16:
If you google for "ARM register arguments", you'll find that 4 first args are passed in registers, the rest on stack. Viper apparently doesn't handle stack args so far.

peterhinch, 2015-07-16:
That may well explain all the results I've been seeing. I'll address this tomorrow and report back. Thanks for your help!

peterhinch, 2015-07-17:
Alas it isn't that simple. The following fails:
```python
@micropython.viper
def get1(src:ptr8, dest: ptr8, arg: int, arg2: int) -> int:
    dest[1] = src[1] + arg + arg2
    return dest[1]

c = bytearray(4)
b = bytearray([0,1,2,3,4])
print(get1(b, c, 4, 5))
print(c)
```
which produces
```
>>> import rats3
0
bytearray(b'\x00\x00\x00\x00')
>>> 
```
Once again we have code which runs but produces incorrect results. If I reduce the number of arguments to three it works. But the issue isn't as simple as a three argument limit. With my actual code I reduced the arguments to two bytearrays and an integer as per the working version of the above test. The outcome was code (tested without the decorator) which ran but produced incorrect results. I have finally managed to get viper code running by passing a single argument, that being the calling object instance. This works whether implemented as a method without explicit arguments or as a plain Python function (with the instance passed as the sole argument): in deference to your suggestion I've settled on the latter. For completeness I've posted the working code below.

After spending much time on this (coding it in assembler would have been quicker) I'm convinced there are subtle problems with argument passing in viper. As a minimum I'd suggest that, if there is an absolute limit on the number of arguments, the compiler should issue an error if it is exceeded. I also think that issuing code which appears to run but produces incorrect results is a bug.

Working code:
```python
@micropython.viper
def _setbuf_data(objEPD): # 5.85S
    line = int(objEPD.line)
    offset = int(objEPD.offset)
    pixelmask = int(objEPD.pixelmask)
    buf = objEPD.linebuf                # Optimisation: use local namespace
    image = objEPD.image
    index = 0                           # odd pixels
    for b in range(BYTES_PER_LINE, 0, -1):
        buf[index] = int(image[offset + b - 1])  ^ pixelmask | 0xaa
        index += 1
                                        # scan line
    scan_pos = (LINES_PER_DISPLAY - line -1) >> 2
    scan_shift = (line & 3) << 1
    for b in range(BYTES_PER_SCAN):
        buf[index] = 3 << scan_shift if scan_pos == b else 0
        index += 1
    for b in range(BYTES_PER_LINE): # Even pixels
        pixels = ((int(image[offset + b])  ^ pixelmask) >> 1) | 0xaa
        pixels = (((pixels & 0xc0) >> 6)
            | ((pixels & 0x30) >> 2)
            | ((pixels & 0x0c) << 2)
            | ((pixels & 0x03) << 6))
        buf[index] = pixels
        index += 1
```

dpgeorge, 2015-07-23:
Sorry for the slow response here.  The issue (fixed by e45c1dbd6f4e313f21862c853d55499447c844e6) is that viper could only take up to 3 arguments.  Anything beyond that it just silently failed and returned 0.  That explains most of your problems.

Now you can pass 4 args so most of the above code should work.  You should also have no problem writing class member functions using viper: the first arg should be `self` as usual and it should just work.

In f17e663493c80e2b9b9dbc8b6340c9239efc92a0 I added code to issue an error when 5 or more args are used in a viper function.

@peterhinch can you please check your code snippets above with the fixes I made and let me know if there are still any outstanding problems.  Everything should either work or raise an error.

peterhinch, 2015-07-24:
@dpgeorge I can confirm that my code samples now produce correct results (or a meaningful error message if too many arguments are passed).

dpgeorge, 2015-07-24:
Ok, thanks, I'll close this then.

General recurring issue here is that viper is not documented.  The best "documentation" at the moment comes from the tests which show exactly what viper can do.

closed 2015-07-24 by dpgeorge

PappaPeppar, 2015-07-24:
I would like to say that the documentation on this topic back at the Kickstarter pages is pretty good.
https://www.kickstarter.com/projects/214379695/micro-python-python-for-microcontrollers/posts/664832
https://www.kickstarter.com/projects/214379695/micro-python-python-for-microcontrollers/posts/665145

Maybe it should be moved to the current wiki.