micropython/micropython #1388: dac1 and dac2 channels interfere with each other

bmentink, 2015-07-21:
Have found an issue with the dac module:

* If I setup dac(1) with a sinewave at 6khz without doing anything to dac(2), output is just fine. 
* If I have dac(1) outputting on the pin then try and display a 63khz triangle on dac(2), the triangle displays but kills dac(1)
... dac(1) sits at some dc level.
* Even with the same frequency used for both calls to dac.write_timed(), I still have the issue.

Code used to test was:
----------------------------------------------------------------------------------------------
# 6.3khz Sine wave
buf = bytearray(1000)
dac1 = DAC(1)
dac1.write_timed(buf, 6300 * 1000, mode=DAC.CIRCULAR)

#Triangle wave 63Khz by default.
buf2 = bytearray(100)
dac2 = DAC(2)
dac2.write_timed(buf2, 63000 * 100, mode=DAC.CIRCULAR)
-------------------------------------------------------------------------------------------------

I have not included the code that fills in the buffers.

dpgeorge, 2015-07-21:
Thanks for the report.  The issue is probably because the timer is reinitialised when you call write_timed the second time and this resets the first DAC.  In principle both DAC channels should be able to run at the same time, even using the same timer.

Related issue is #1129.

bmentink, 2015-07-21:
From a cursory inspection, the issue seems to be in pyb_dac_write_timed(), I think the same dma channel is used for both ..

bmentink, 2015-07-21:
I think the problem is that both the DMA device and the DAC device are reset each time the function pyb_dac_write_timed() is called, the resets should be done once and the dac channel setup and stopped/started for each instance ... maybe someone else who wrote the code could have more input ..

dpgeorge, 2015-07-21:
Should now be fixed.  Patch linked above allows you to specify a Timer object instead of frequency.  You can do something like this:
```python
dac1.write_timed(buf1, Timer(6, freq=6300*1000), mode=DAC.CIRCULAR)
dac2.write_timed(buf2, Timer(7, freq=63000*100), mode=DAC.CIRCULAR)
```
Or even using the same timer if the freq is the same:
```python
tim6 = Timer(6, freq=6300000)
dac1.write_timed(buf1, tim6, mode=DAC.CIRCULAR)
dac2.write_timed(buf2, tim6, mode=DAC.CIRCULAR)
```

I have verified (with an oscilloscope) that it works.  @bmentink can you also confirm it works for you, then we can close the issue?

bmentink, 2015-07-21:
That was quick :) 
I tried, but now I have things swapped. My sinewave displays (dac1), but dac2 the triangle does not.
I used the different timer method.

bmentink, 2015-07-21:
The 2nd method leaves me the same as before: Triangle working (dac2) Sine not (dac1)

bmentink, 2015-07-22:
Found the issue for me. I had a 1000 byte buffer for one of my waveforms (dac1), changed it back to 100, and everything works with the dual timer system.

I had 1000bytes for dac1 and 100 bytes for dac2,  as with the same timer I wanted difference in frequencies but synced .. looks like this is not possible.

peterhinch, 2015-07-22:
This is a great enhancement! The link to timer 6 was a major limitation. Can the ADC's also be set up to use specific timers?

bmentink, 2015-07-22:
Works great, tested for a couple hours ... many thanks for acting so quickly on this.

dpgeorge, 2015-07-22:
@bmentink do you have more info about your problem with large buffers?  Does 1000 bytes work for 1 channel at a time?  Does it work with 1000 bytes for both dac1 and dac2 at the same time?  What about 1000 and 1100 for the 2 channels at the same time?

bmentink, 2015-07-22:
Yes, 1000 bytes for 1 channel only works. Both at 1000 does not. 1000 & 1100 also does not.

I suspect that at the frequency I am going at (6.4khz) 500 byte buffer is about the limit to shift every timer cycle .. timer would be 6400 * 500 = 3.2Mhz, the other channel is the same freq 64000 * 50 = 3.2Mhz  ....should it be capable of more than 3Mhz?

bmentink, 2015-07-22:
Did another experiment with setup like this:
----------------------------------------------------------------
sample = 100
buf = bytearray(sample*10) # 10 x freq difference.
buf2 = bytearray(sample*10)
tim6 = Timer(6, freq=49300*sample)
dac1.write_timed(buf, tim6, mode=DAC.CIRCULAR)
dac2.write_timed(buf, tim6, mode=DAC.CIRCULAR)
----------------------------------------------------------------

About 49000*1000 is where it dies ... 48000*1000 freq works.

dpgeorge, 2015-07-23:
@bmentink I think you're just at the limit of the DACs capabilities in terms of high speed output.  In the datasheet it says it takes around 3us for the output to settle when changing from maximum to minimum (ie full swing).  So that means you can do at most a 333kHz/2 square wave.  If you're doing a sine wave or other waveform that doesn't have sharp changes then you can probably clock closer to 1MHz.

In your example above with freq=4.8MHz I would say this is really at the DACs limit.  And note that Timer(6) is 16-bit so it can't de precise frequencies at high frequencies.  Eg freq=4800000 is actually 4941176 and the next step up is 4950000 which gives an actual frequency of 5250000.

The original issue here is now fixed, so will close the issue.  If you find a *software* problem please open another ticket with precise details.

closed 2015-07-23 by dpgeorge