micropython/micropython #1390: Partial string hex escape: Assertion `0' failed

ampli, 2015-07-23:
```
$ micropython 
Micro Python v1.4.4-126-gf17e663-dirty on 2015-07-23; linux version
>>> "\x0"
micropython: ../py/lexer.c:458: mp_lexer_next_token_into: Assertion `0' failed.
```

dhylands, 2015-07-23:
That seems to be invalid in python2:
```
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> "\x0"
ValueError: invalid \x escape
```
and in python3:
```
Python 3.4.0 (default, Jun 19 2015, 14:20:21) 
[GCC 4.8.2] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> "\x0"
  File "<stdin>", line 1
SyntaxError: (unicode error) 'unicodeescape' codec can't decode bytes in position 0-2: truncated \xXX escape
```

dhylands, 2015-07-23:
Or were you trying to say that it should throw an exception and not assert?

ampli, 2015-07-23:
Yes. Such errors shouldn't cause an assert.

dpgeorge, 2015-07-23:
Ack.  Just need to think of a neat way to handle it.  It's not a common thing that will happen so just needs to emit a basic SyntaxError.

dpgeorge, 2015-07-23:
Thanks for bug report, not fixed.

closed 2015-07-23 by dpgeorge