micropython/micropython #1393: Device using the block protocol is unreliable

peterhinch, 2015-07-25:
I have been experiencing random behaviour including crashes when implementing a device using the block protocol. It's proved difficult to write a script to yield a repeatable failure and my best effort is unfortunately a little long. However it does produce a repeatable crash. I suspect a memory issue.

The code and test data is at [https://github.com/peterhinch/blockdevice.git]

peterhinch, 2015-07-26:
In case anyone has already cloned this, I've now included a second test.

dpgeorge, 2015-07-27:
I pushed a fix for this which hopefully actually fixes it!  The problem was that the structure holding the mount info was not scanned during a GC so was being reclaimed.  Please check latest code with your example.

peterhinch, 2015-07-28:
My tests now run fine!

closed 2015-07-28 by dpgeorge