micropython/micropython #1395: Add support for STM32F7

dhylands, 2015-07-28:
I'm in the process of adding support for the STM32F7 Discovery board.

I've created one commit for adding the hal and cmsis components:
https://github.com/dhylands/micropython/commit/750da8e0a88491d4959c55b1972651798158c438

and a second commit for getting micropython to the REPL (whatever the top commit on this branch is)
https://github.com/dhylands/micropython/commits/add-stm32f7-support

It isn't ready for a PR yet, but I thought I would mention what I'm working on. I have micropython booting to a HWUART REPL.

The filesystem doesn't seem to be working
USB doesn't seem to be working
The I2C and SPI are currently hacked off (this is the major reason for not creating a PR yet).
The HAL_I2C_Init changed the way the clock frequency calculation is done, which is a pain.

Most of the changes were to change #include <stm32f4xx_hal.h> to be #include STM32_HAL_H where STM32_HAL_H is defined to be stm32f4cc_hal.h or stm32f7xx_hal.h

I added a capability to add board C files, and have a board early initialization routine that can be called. This is ueful for setting up board specific gpios without cluttering up main.c I'm using on STM32F7DISC to turn off the backlight. I plan on removing the #if 0'd NETDUINO stuff and moving that to a NETDUINO specific board initialization file.

And I added -DSTM32$(MCU_SERIES) to CFLAGS, so that the few places that there are differences that they can be dealt with.

dpgeorge, 2015-07-28:
Sounds great @dhylands!

To speed up merging this, we should try to break things into clean, functional pieces.  I'm happy to have it working just to the extent of a UART REPL to begin with, then other features can come later.

Re $(MCU_SERIES): probably should (automatically) make an upper case version as well so that upper case macros are consistently upper case.

We need to see how much of the startup_xx.s, xx_it.c and system_xx.c files are common across MCU series.  Probably these files should go in boards/ along with .ld files and _af.csv files.  I did consider adding an mcu/ directory within stmhal/ to put all these files, but thought that having them in boards/ is just as good.

Re the f7 hal files: to reduce the code size of this repo I only added .h and .c files that we actually use.  We are going to have an ongoing problem with the size of the hal directory (and the entire git repo), especially if we upgrade the hal (since then git stores the entire history of all hal versions we used...).

> I also discovered that there appears to be no way to enter DFU mode (on the STM32F7) without using a HW switch

You mean pyb.bootloader() doesn't work?  I'd think that you should always be able to jump to the right address to enter DFU mode.

dhylands, 2015-07-28:
> We need to see how much of the startup_xx.s, xx_it.c and system_xx.c files are common across MCU series. Probably these files should go in boards/ along with .ld files and _af.csv files. I did [...]

I was able to start the _it.c file. If we rename the startup_xx.s to startup_xx.S (so that it's preprocessed) then we'll be able to share that as well.

I can put the i2c and spi stuff behind #if !defined(STM32F7) macros to allow it be merged.

So I'll start with a PR for just the hal. The initial commit will just be virgin files with unix line endings, and just the files that are already in the f4.

dpgeorge, 2015-07-29:
Done in 7c934ae5019d285b58cec3d1edcbc44d90fa6295 up to 92d4b51ad5d828930334f87d9619a78b5877a384.

closed 2015-07-29 by dpgeorge