micropython/micropython #1405: Inconsistent pin behaviour

dhylands, 2015-08-02:
I've run into an issue which I throught I would bring up, and see how to best go about solving it.
If I do:
```python
pin = pyb.Pin('J12', pyb.Pin.IN)
```
Then I get a pin which has been run through the Pin type's make_new and init_helper function. If I do:
```python
pin = pyb.Pin('J12')
```
Then I get a pin which has been run through the Pin type's make_new function, but not the init helper function. If I then try to query the pin's value, it always returns zero because the clock for that port hasn't been initialized. I can move the clock initialization to the make_new function to get around this.

However, I can also get a pin via:
```python
pin = pyb.Pin.cpu.J12
```
and I get a pin from the dictionary in flash that hasn't been run through make_new and have the same problem. I can query the pin:
```python
>>> pyb.Pin.cpu.J12
Pin(Pin.cpu.J12, mode=Pin.IN)
>>> pyb.Pin.cpu.J12.value()
0
>>> pin = pyb.Pin('J12', pyb.Pin.IN)
>>> pyb.Pin.cpu.J12.value()
1
```
For ports A-D, you can do the above and the first call to value() will return 1 because the clock is enabled.

My question is, do we want to resolve this inconsistency?
Should we consider raising an error on a pin that hasn't been initialized?
Or should we enable the clock?

Right now, when you ask for pyb.Pin.cpu.J12 you're getting a pin from a local_dict:
https://github.com/micropython/micropython/blob/master/stmhal/pin_named_pins.c#L43

By changing this to an attr and we could have the gpio clock enabled. Or we could throw an error if init hasn't been called yet, or we could do nothing.

So I just thought I'd throw this out for discussion.

danicampora, 2015-08-02:
This is a good one. The CC3200 has the same issue, but the outcome is much worse, if you try to access a pin which has no clock yet, guess what? Hardfault...

dpgeorge, 2015-08-02:
There is a more important thing to resolve first: Pin.init() on pyboard is different to Pin.init() on wipy.  They must be the same, else it's just crazy.  Eg the 1-wire driver currently does not work on wipy, and there's no reason it shouldn't because it uses simple bit-banging.

danicampora, 2015-08-02:
> There is a more important thing to resolve first: Pin.init() on pyboard is different to Pin.init() on wipy.

Well,  `Pin.init()` on both boards were written to mirror the internal details of their respective microcontroolers. In order to not break existing code I guess that I would have to rewrite `Pin.init()` in the WiPy to be the same as in the pyboard. I'll add this to my TODO list.

dpgeorge, 2015-08-02:
> In order to not break existing code I guess that I would have to rewrite Pin.init() in the WiPy to be the same as in the pyboard.

We should take this opportunity (with 2 quite different MCUs) to think of a better common-denominator Pin class.  Eg we can have Pin.OUT instead of Pin.OUT_PP (since the default being push-pull is what people expect).  If we want to have a nice API for all peripherals (and that's been the underlying theme to a lot of discussion so far) then we may need to make some backwards-incompatible changes to pyboard.

danicampora, 2015-08-02:
> If we want to have a nice API for all peripherals (and that's been the underlying theme to a lot of discussion so far) then we may need to make some backwards-incompatible changes to pyboard.

Cool, if you are OK with that then we should come up with a better constructor for the Pin class. I think that stmhal's constructor is fine, except that we should decouple the mode (IN or OUT) from the pin type (PP or OD).

Here's my take:

```python
Pin(pin, mode, *, type=Pin.PP, pull=None, af=0)
```
`mode` can be `IN` or `OUT`
`type` can be `PP` or `OD`
`pull` can be `PULL_UP`, `PULL_DOWN` or `None` (or `PULL_NONE` but we could save on QSTRs)

Alternate functions can vary widely from micro to micro so I think it's OK for this last param to be port specific.

Thoughts?

danicampora, 2015-08-02:
By the way, I am 100% on making the WiPy and the pyboard API compatible as much as possible. See https://github.com/micropython/micropython/commit/c6926c374dc0408d7ef8fa6385207430c122e6cf ;-)

dhylands, 2015-08-02:
PP and OD only apply to output style pins (either OUT or AF)
If we decide to add type, we can still have OUT_OD and OUT_PP for backwards compatability.
mode should be one of IN, OUT, or AF

I think breaking it down like that makes it more portable to other MCUs, which can then support whichever combinations of those parameters makes sense.

The original issue I wanted to bring up was what should happen when you try and use an uninitialized pin? Or should such a thing even exist? All of the MCUs that I'm aware of default to pins being inputs with no pullups/pulldowns by default. So when you ask for pyb.Pin.cpu.J12 should the pin come in a working state? (implied input).

danicampora, 2015-08-02:
> The original issue I wanted to bring up was what should happen when you try and use an uninitialized pin? 

For me the user should be careful to configure pins and other hardware before accessing it. What happens in the pyboard if you try to use an UART without initializing it first?

```python
import pyb
pyb.UART(2).write('abc')
```
?

If you do that with the CC3200 it will hard fault, because the UART is not clocked yet. I don't think we should protect against this, stuff needs to be initialized properly before being used, period.

dpgeorge, 2015-08-02:
> What happens in the pyboard if you try to use an UART without initializing it first?

It raises an OSError exception (with EBUSY code).

danicampora, 2015-08-02:
> It raises an OSError exception (with EBUSY code).

Well that's obviously nicer than a hard fault, but having to check everywhere if a peripheral is initialized will end up adding quite some lines of code.

dpgeorge, 2015-08-02:
The check is done by ST's HAL.  It checks for a few things (like initialsed peripheral, correct params, timeout, etc) and if it returns an error then we just raise that as a Python exception.

dhylands, 2015-08-02:
OK. I think it might be useful to print out the inialized status in the
print function (especially if the info is available). For the pins we
mostly bypass the hal.

dhylands, 2015-08-03:
I realized that I can't do anything with the pin objects because we never actually allocate anything from the heap. The returned pin is always a flash object.

dhylands, 2015-08-03:
Perhaps the best thing to do is to clarify the various examples in the documentation.

danicampora, 2015-08-03:
> Perhaps the best thing to do is to clarify the various examples in the documentation

For instance?

dhylands, 2015-08-03:
Currently,
pyb.Pin.cpu.A1 is an uninitialized pin
pyb.Pin('A1') is an uninitialized pin
pyb.Pin('A1', pyb.Pin.IN) is an initialized pin

This is consistent with:
pyb.UART(1) being uninitialized
pyb.UART(1, 9600) is initialized

danicampora, 2015-08-03:
Discussion related to the `Pin.init()` moved to #1408

jonnor, 2024-07-02:
Is this still relevant? I believe considerable consistency improvements have been done to the Pin API over the last 9 years.

dpgeorge, 2024-07-03:
> I believe considerable consistency improvements have been done to the Pin API over the last 9 years.

Indeed, a lot of time has passed and the API is pretty stable now.  Any further improvements can have separate issues for them.

closed 2024-07-03 by dpgeorge