micropython/micropython #1408: Consistent Pin API for all ports

danicampora, 2015-08-03:
This continues what started being discussed in #1405. The Pin API as far as the constructor (and the init()) method is concerned needs to be modified a bit in order to make it more general and suitable for a wide range of MCUs. Here's my proposal:

```python
Pin(pin, mode, *, type=Pin.PP, pull=None, af=0)
```
`mode` can be `IN`, `OUT` or `AF`
`type` can be `PP` or `OD`. Irrelevant when `mode=Pin.IN`.
`pull` can be `PULL_UP`, `PULL_DOWN` or `None` (or `PULL_NONE` but we could save on QSTRs)
`af` is only relevant when `mode=Pin.AF`. Alternate functions can vary widely from micro to micro so I think it's OK for the values of this param to be port specific.

dhylands, 2015-08-03:
That looks reasonable to me, and if we decide to go forward with this, I'm happy to change the stmhal stuff to implement this (and keep the older stuff for backwards compatability - perhaps with an ifdef for size considerations - so ifdef on).

danicampora, 2015-08-04:
Cool, let's wait a bit for @dpgeorge 's opinion.

dpgeorge, 2015-08-05:
I'd rather not have PP and OD separate.  I think it's simpler to have just `mode` and `pull` (ignore for now alternate function).  Then `mode` would be one of: `IN`, `OUT` or `OUT_OD`.  In this case `OUT` would be push-pull (and we provide `OUT_PP` as an alias on pyboard for legacy support).  Or, `IN`, `OUT` and `OD`.

There may also be an analog mode for some MCUs (eg `mode=Pin.ANALOG`); but this could be encompased by ADC setup and/or alternate-function.

There are other things we can support:
- pull strength (available on ESP8266)
- drive strength (available on CC3200)
- speed (available on STM)

To support pull strength, those chips with it can have `PULL_DOWN` as the default strength, and provide `PULL_DOWN_4K` or whatever for specific strengths (or use STRONG, WEAK etc).

But maybe this is overcomplicating it.  Most people will only ever use direction and pull up/down.

dhylands, 2015-08-05:
drive strengths and pull strengths start to become more important when the speeds start increasing, and trace lengths start getting longer. When I was working on phone chips we'd often have to increase both on the dev boards (which were way bigger than a phone) and reduce them (for power considerations) on a real phone (which often have 10 layer boards to get all of the traces routed).

So I like having PULL_DOWN just be a "default" and let the board config decide what's appropriate for the default.

I think having OUT imply OUT_PP sounds reasonable (I'd do the same for AF). OD usage is alot less common (bit banged i2c and speciality power considerations are typically where this is used). I guess you could also use it for interfacing with odd voltages (i.e. have an external pullup to the desired voltage).

dpgeorge, 2015-08-06:
I think OD is very useful when it's available, so should certainly be an option.

Do we need to support both AF_OD and AF_PP (with AF=AF_PP)?

dhylands, 2015-08-06:
AF_OD is useful for the voltage conversion thing. If you were interfacing with a 1.8v part that wasn't 3.3v tolerant (for example) then you could use AF_OD with external pullups that pullup to 1.8v, or you were interfacing with a 5v part that didn't properly recognize 3.3v as a high (I've done that before). Using AF_OD on the PWM would allow the voltage translation chip to be eliminated (for example, this H-Bridge chip https://www.pololu.com/product/712/specs needs 4v to get a proper logic high)

danicampora, 2015-08-06:
> Do we need to support both AF_OD and AF_PP (with AF=AF_PP)?

We do. For instance, the CC3200 needs to configure the pins with `mode=AF` and `type=OD` for I2C operation, otherwise it doesn't work (simply assigning the AF to SDA or SCL is not enough).

> I'd rather not have PP and OD separate. I think it's simpler to have just mode and pull

Yes, it is simpler, but less flexible and will be harder to map to other MCUs in the future. I'd prefer to have it separate, even if it doesn't make much sense for the pyboard (but it makes sense for the CC3200).

> There are other things we can support:
>    - pull strength (available on ESP8266)
>    - drive strength (available on CC3200)
>    - speed (available on STM)

We could do two things:
1. Leave this extra options out of the basic API and let each port provide the ones it needs.
2. Make them part of the basic API and simply ignore them if not supported.

dpgeorge, 2015-08-06:
Ok, we need PP & OD for AF mode.

> Yes, it is simpler, but less flexible and will be harder to map to other MCUs in the future.

I'm not convinced it's less flexible.  Both cc3200 and stmhal are pretty much the same when configuring GPIO: they have a general mode (IN, OUT, AF, ANALOG) and then 1 bit to select PP or OD, 1 bit for PULL_UP and 1 bit for PULL_DOWN.  cc3200 has an extra 2 bits for drive strength, and stmhal has bits for io speed.

We can do it 3 different ways:

1. All in one: mode= one of IN, IN_PU, IN_PD, OUT, OUT_OD, AF, AF_OD, ... (this is how Espruino does it, you probably also want OUT_OD_PU, AF_OD_PU, etc).

2. All separated: mode= IN, OUT, AF, ANALOG; drive= PP, OD; pull= NONE, UP, DOWN (then strength=...; af=...; etc).

3. Hybrid of above: mode= IN, OUT, OUT_OD, AF, AF_OD, ANALOG; pull= NONE, UP, DOWN (then strength=...; af=...; etc).

[Note that mbed has 2 config values: direction= INPUT or OUTPUT, and type= PULL_NONE, PULL_UP, PULL_DOWN and OPEN_DRAIN (so no OD with PU).]

We want either 2 or 3, since they are the most flexible, and I'd say they both have the same flexibility.

One advantage(?) of 2 is that the `drive` parameter can be extended to include the drive strength when an MCU supports that.  Eg `drive=PP_2MA` for cc3200.  Then it doesn't need the `strength` parameter.  `drive=PP` would give the default strength.  (Is strength relevant for OD?  If so then you'll need to do, eg, `drive=OD_4MA`.)

Ultimately we want the following features:

1. Base-line config is simple, easy, short, common across MCUs and for the most case all you'll need.  Eg `pin = Pin('X', Pin.OUT)` or `pin = Pin('Y', Pin.IN, Pin.PULL_UP)`.  Then code like the onewire driver will work on all uPy boards.

2. Easy path for individual ports to support more sophisticated features such as AF, drive strength

danicampora, 2015-08-06:
> All in one: mode= one of IN, IN_PU, IN_PD, OUT, OUT_OD, AF, AF_OD, ... (this is how Espruino does it, you probably also want OUT_OD_PU, AF_OD_PU, etc).

This is crazy for me, hard to read and will lead to like a hundred values to cover all the possible combinations. The fact that others do it doesn't mean that it's right. For instance, the Arduino Pin API is extremely simple to the point that it sucks (mode: INPUT, OUTPUT, or INPUT_PULLUP, that's it, pff...).

> [Note that mbed has 2 config values: direction= INPUT or OUTPUT, and type= PULL_NONE, PULL_UP, PULL_DOWN and OPEN_DRAIN (so no OD with PU).]

This is absurd. It's perfectly fine (and logical) to want OD with a PU.

> Ultimately we want the following features:

>  - Base-line config is simple, easy, short, common across MCUs and for the most case all you'll need. Eg pin = Pin('X', Pin.OUT) or pin = Pin('Y', Pin.IN, Pin.PULL_UP). Then code like the onewire driver will work on all uPy boards. 

> - Easy path for individual ports to support more sophisticated features such as AF, drive strength

I agree with both.

> One advantage(?) of 2 is that the drive parameter can be extended to include the drive strength when an MCU supports that.

The problem with that is, that if a MCU doesn't support drive strengths and a driver library in Python uses `drive=PP_2MA` it will fail. If instead the strength is passed separately it can simply ignore it and continue.

I vote for either 2 or 3, but I think we should keep the special values (drive strength, pull strength, speed, etc) passed as separate params.

danicampora, 2015-08-06:
>  (Is strength relevant for OD? If so then you'll need to do, eg, drive=OD_4MA.)

I don't think drive strength is relevant for OD.

blmorris, 2015-08-06:
Does anyone know if drive strength and (signal) speed can be mapped to each other? To me it looks like they are different ways to specify driver output impedance: reducing the impedance increases the drive strength and increases the signal transition speed. It also increases the power required.
If we can determine that the likely target architectures support at most one of [speed, drive strength, output impedance] then we can safely assume that it really is just different names for the same parameter. In this case we could either choose a common name that can be shared across ports, or accept all names and map them to the same underlying parameter (if it exists for that port) for compatibility across ports.

danicampora, 2015-08-06:
I'll investigate.

blmorris, 2015-08-06:
One architecture that I have some experience with is the Analog Devices Blackfin. The data sheets for these devices provide detailed plots of current vs pin voltage and rise/fall times vs load capacitance for each driver type, even plotting for different pin supply voltages (1.8, 2.5, 3.3V). It all amounts to a detailed specification of output impedance.
For the STM32F405 (pyboard processor) the corresponding information can be found in 'Table 50. I/O AC characteristics' of http://www.st.com/web/en/resource/technical/document/datasheet/DM00037051.pdf
There are 2 OSPEEDRy bits; for each of the four available output configurations the table lists maximum frequency and maximum rise/fall times for different values of load capacitance and VDD. Again, this looks like a specification of output impedance, for which we have four possible choices.
It seems to me that for drive strength (speed, impedance, etc) we are likely choosing from at most four values; each port can assign names to these values (i.e. low = 0, med = 1, fast = 2, high = 3) but this doesn't preclude passing integer values for script portability across ports. A port that doesn't support drive strength (cc3200?) can just ignore the parameter.

danicampora, 2015-08-06:
Interesting. The CC3200 does support drive strength.

blmorris, 2015-08-06:
@danicampora -
>Interesting. The CC3200 does support drive strength.

Sorry, I did a quick scan of the cc3200 data sheet and missed it. I see it now; options are 2, 4, 8, and 14mA; strength is implemented via 3 bits, 14mA is achieved by activating all of the drivers:
2mA + 4mA + 8mA = 14mA
The specified rise and fall times for the cc3200 aren't a perfect match for the pyboard, but probably close enough for this API to treat them as the same.

danicampora, 2015-08-06:
@blmorris Cool! Thanks for bringing this. Looks like we can have one param to cover both drive strength and speed.

dhylands, 2015-08-06:
Some processors also support slew rate. The places where you see this show up first is on high speed memory interfaces, and then on the higher speed processors (> 1 GHz) I'm pretty sure that I've seen both drive strength and slew rate on gpio pins.

dhylands, 2015-08-06:
Heh - the processor used with the teensy (MK20DX256)  supports slew rate (fast/slow) in addition to drive strength (low/high).

blmorris, 2015-08-06:
>  Looks like we can have one param to cover both drive strength and speed.

Yes, I think so. Thinking a bit more, those relatively rare situations where you really have to be careful to get drive strength right are situations where you are most likely paying close attention to the specs of a particular processor family and aren't that concerned with script portability; therefore I wouldn't worry too much about the drive strength parameters not being a perfect match across families.

@dhylands 
> Heh - the processor used with the teensy (MK20DX256) supports slew rate (fast/slow) in addition to drive strength (low/high).

I was about to answe that I didn't think we needed to worry about slew rate yet… no such luck.
Looking at the MK20DX256 data sheet, there are two drive strengths (high and low) and two slew settings (disabled for fast transitions, enabled to slow down the transitions).
A quick-and-dirty way to support the teensy using the a single strength/speed parameter would be to map bit 0 to (~ slew) and bit 1 to strength; this would have to be documented specifically for the teensy of course. Another option would be to future-proof for situations where MicroPython is routinely running on much faster processors by implementing a slew parameter now which would get ignored by all ports except the teensy; somehow this strikes me as overkill, but maybe not.

dhylands, 2015-08-06:
I don't know of a particular application where slew would be important. I'd be happy to incorporate it into the strength.

blmorris, 2015-08-06:
> I don't know of a particular application where slew would be important.

Mostly high-speed parallel data interfaces, the kind where your need to do board trace-level simulation to get it right, I think. Otherwise, it is often part of the spec for serial line drivers - IIRC, RS232 drivers are supposed to be slew-rate limited to reduce cross-talk in long unshielded cable runs. I'm surprised to see it in the MK20DX256, but maybe it is intended to allow it to drive long unshielded serial lines directly. If this is useful, it would at least be important to note that for the teensy even-numbered drive strengths are slew-limited.

danicampora, 2015-08-11:
OK so I think it could go like:

```python
Pin(pin, mode, *, pull=None, af=0, drive_str=Pin.MED_STR)

- `mode` can be `IN`, `OUT` (default), 'OUT_OD`, `AF` or 'AF_OD`
- `pull` can be `PULL_UP`, `PULL_DOWN` or `None` (default).
- `af` being port specific.
- `drive_str` can be `LOW_STR`, MED_STR` or `HIGH_STR`. Do we want/need more granularity?
```

But I think that maybe `drive_str` (and slew rate, and others...) doesn't need to be part of the basic API, and then only the ports that support such features are free to provide them. Like @blmorris said, when you need to worry about things that are so platform specific, script portability is less important.

pramasoul, 2015-08-11:
Perhaps a <code>Pin</code> should default to be an input, for circuit safety, rather than an output.

danicampora, 2015-08-11:
> Perhaps a Pin should default to be an input, for circuit safety, rather than an output.

Good point, but the `mode` is a required param so no need to specify a default value. My mistake, now corrected ;-).

dhylands, 2015-08-11:
When declaring a pin as an output, I think it makes sense to provide an initial value for the pin, so that it doesn't "glitch". So I'd propose that we add a value=0/1 argument.

dpgeorge, 2015-08-11:
I'm happy with @danicampora's suggestion just above.  Except that `drive_str` is not the best name (str means string in Python).  It could be just `drive`, or `strength` or `out_power`.

dhylands, 2015-08-11:
I think you could also shorten drive_strength to either just drive or strength. I don't see any ambiguity for either one.

chrismas9, 2015-08-11:
I prefer drive because it covers drive strength (current limit) and drive speed (slew rate). An instance of where slew rate important is when you have a lot of simultaneous switching outputs you want low slew rate, but you would want high slew rate on SPI if you were running it at say 20MHz. Some chips limit the number of fast simultaneous switching outputs per ground pin to prevent ground bounce which can crash it.

blmorris, 2015-08-11:
re: drive_strength - STM uses 'speed' for this parameter, which I found confusing coming to this architecture given that it has nothing to do with clock speed. The EE in me likes 'impedance', but it is probably a non-starter for two reasons: First, it would be defined as reciprocal to 'strength'; low impedance <--> high strength. Second, it would only make sense to EE's and not to most software developers ;)
I'm happy with either `drive` or `strength` being `LOW`, `MED`, `HIGH`, so long as it can also be specified directly as an integer. That way we don't need to specify finer granularity in the API but it can still be accessed if the architecture offers it. We can talk about `slew` later when it becomes an issue, and let @dhylands decide if / how to implement it on the teensy.
@chrismas9 - agreed; and I appreciate the details regarding slew rate.

danicampora, 2015-08-11:
Ok I like `drive` too. I think that adding the optional `value` param to the constructor is also a good idea. Does everyone agree?

dpgeorge, 2015-08-11:
So `drive` means whatever a given port defines it to mean (something to do with strength)?  And all ports provide `LOW_DRIVE`, `MED_DRIVE` and `HIGH_DRIVE` constants?

blmorris, 2015-08-11:
`drive` is good, but it just occurs to me that `drive = Pin.LOW` or `drive = Pin.HIGH` _might_ be confused for specifying a level. `drive_current` avoids the ambiguity entirely, if we think it is worth a few extra bytes / keystrokes. Maybe `drive_cur`? Or resolve it by @dpgeorge suggestion of naming the constant `LOW_DRIVE` instead?

danicampora, 2015-08-11:
> So drive means whatever a given port defines it to mean (something to do with strength)? And all ports provide LOW_DRIVE, MED_DRIVE and HIGH_DRIVE constants?

Yes.

pramasoul, 2015-08-12:
An open-drain is a LOW_DRIVE and HIGH_FLOAT output, and a push-pull is LOW_DRIVE and HIGH_DRIVE ;-). We might look for words more connected with force or speed, like SOFT, MODERATE, HARD, or EASE, PUSH, SLAM.

danicampora, 2015-08-13:
> An open-drain is a LOW_DRIVE and HIGH_FLOAT output, and a push-pull is LOW_DRIVE and HIGH_DRIVE ;-). We might look for words more connected with force or speed, like SOFT, MODERATE, HARD, or EASE, PUSH, SLAM.

Well, finding the perfect name is almost impossible... what about naming the `drive` constants `LOW_STRENGTH`, `MED_STRENGTH` and `HIGH_STRENGTH` ? or too long ?

pfalcon, 2015-09-13:
Should this be closed in favor of #1430?

danicampora, 2015-09-13:
Yes, thanks.

closed 2015-09-13 by danicampora