micropython/micropython #1418: Wrong internal ADC values

iabdalkader, 2015-08-06:
When reading VREF/TEMP after VBAT channel is configured (on STM32F42x) I'm getting unexpected values, from the datasheet:

>Note: On STM32F42x and STM32F43x devices, VBATE must be disabled when TSVREFE is
set. If both bits are set, only the VBAT conversion is performed.

I suggest clearing VBATE and TSVREFE after reading channels in `adc_read_core_*` the vbat, vref and temp channels are re-configured each time `adc_read_core_*` is called anyway.

A different but related issue, those internal channels need more sampling time (minimum 10us) there should be checks in `adc.c` for that, something like:
```
if (adc_obj->channel == ADC_CHANNEL_TEMPSENSOR ||
adc_obj->channel == ADC_CHANNEL_VREFINT ||
adc_obj->channel == ADC_CHANNEL_VBAT) {
    sConfig.SamplingTime = ADC_SAMPLETIME_480CYCLES;
} else {
  sConfig.SamplingTime = ADC_SAMPLETIME_15CYCLES;
}
```

Note, using 15CYCLES the temp reports about 10 deg higher!

chuckbook, 2016-12-02:
@dpgeorge Indeed the temp. readings will benefit from longer sampling times but not to the extend mentioned by @iabdalkader. We choose to go with 15 cycles for all the internal ADC channels as the most important channel (for us) is VBAT and the voltage divider puts quite some load on the backup battery, so it is better to keep sample times short.
This was the reason to disable VBATE as soon as possible by patching HAL. That went down the drain with the recent HAL replacement. Please take my apologies.
Having a dedicated sample time for Vtemp would be good but only if no side effects are introduced.

iabdalkader, 2016-12-04:
It took some time but I finally found it buried deep in the datasheet:
[image]

You need at least 10us when reading the internal temperature sensor, if you convert that to ADC cycles you'll have the minimum sampling time (which may not be 480 cycles).

I guess I should close this issue now ? or feel free to close it when it's resolved.

dpgeorge, 2016-12-05:
> Having a dedicated sample time for Vtemp would be good but only if no side effects are introduced.

I'm happy to do this.  I just ran some tests using the 8 possible sampling times (from 3 to 480 cycles) and couldn't see any significant difference in the measured core temperature value.  The downside of using a large sampling time is just that it takes a while, but if the datasheet suggests 10us then I guess we need to use 480 cycles.

It really depends on the use of the temp sensor: if it's good for accurate measurements then large sampling time makes sense.  But if it's just used for basic, low-accuracy measurements then small sampling time is ok.