micropython/micropython #1419: REPL problem with quotes

jdolarte, 2015-08-08:
A string containing an open parenthesis and  a quote, REPL kept looking for matching pairs:
```
Micro Python v1.4.4-182-g872f9af on 2015-07-09; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> "\"("                                                                       
...           
```
and also there is a problem with the functions that contain incomplete quotes
```                                                                  
>>> print("\"")                                                                 
...
>>> open('file', 'w').write('\'')                                               
...
```
last one worked fine before 28596edf074f227ff5897ca280d77d38cccf4c85

dpgeorge, 2015-08-08:
The issue is escaped quotes being recognised as normal quotes.

We need to write some tests to check line continuation in the REPL.

alex-march, 2015-09-20:
I was hoping github to close the issue via the commit message saying "Fixes", doesn't it usually happen this way?

dpgeorge, 2015-09-20:
Nope, the issue needs to be explicitly closed.

commit bfb272b9 (2015-09-19): py/repl: Treat escaped quotes correctly in REPL continuation.

Escaped quotes are now recognised correctly in REPL when used
inside normal quotes.

Fixes: #1419
closed 2015-09-20 by dpgeorge