micropython/micropython #1420: sys.implementation() returns wrgon version

ha5di, 2015-08-08:
On Linux v1.4.4 sys.implementation() returns

(name='micropython', version=(0, 0, 1))

instead of

dpgeorge, 2015-08-08:
Most likely this is because your git clone of the repo does not include tags.  Try something like `git pull --tags`.

ha5di, 2015-08-09:
I downloaded source from the release page expecting it is the most official tarball :)

dhylands, 2015-08-09:
The actual version information is generated by querying git. When you use a tarball, then the git information is missing. The official source comes from the git repository, and the tarball is as faithful a representation as a tarball can be, but it doesn't contain the git version information.

ha5di, 2015-08-09:
Understand and I have no problem to build from git, but I think the release tarball must provide proper version. It is a RELEASE :)

pfalcon, 2015-08-09:
Release is a tag in git repository. Tarball is provided as a convenience feature for people who can't use git in a particular setting, with a suggestion to use git on a first occasion ;-).

ha5di, 2015-08-09:
OK, lets close it now. Will build 1.4.5 from GIT :)

dpgeorge, 2015-08-11:
Version should now be correct, even if you didn't pull the repo.  The feature was almost there before, since if you didn't have git installed at all then it would get the version number from the docs.  Now it will also use the docs version number if git exists but there's not git repo.

ha5di, 2015-08-11:
Building 1.4.5 release tarball get this error:

# ./micropython.build
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/qstrdefs.generated.h
Traceback (most recent call last):
  File "../py/makeversionhdr.py", line 107, in <module>
    make_version_header(sys.argv[1])
  File "../py/makeversionhdr.py", line 73, in make_version_header
    info = get_version_info_from_git()
  File "../py/makeversionhdr.py", line 26, in get_version_info_from_git
    if er.args[0] == 128:
IndexError: tuple index out of range
../py/py.mk:132: recipe for target 'build/genhdr/mpversion.h' failed
make: *** [build/genhdr/mpversion.h] Error 1

dpgeorge, 2015-08-11:
Strange. Can you put the following line before the line that fails:
```
print(repr(er))
```
and report the output?

ha5di, 2015-08-12:
Maybe put to wrong place, I added to py/makeversionhdr.py:

if __name__ == "__main__":
    print(repr(er))
    make_version_header(sys.argv[1])

Result:

# make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Traceback (most recent call last):
  File "../py/makeversionhdr.py", line 107, in <module>
    print(repr(er))
NameError: name 'er' is not defined
../py/py.mk:132: recipe for target 'build/genhdr/mpversion.h' failed
make: *** [build/genhdr/mpversion.h] Error 1

dhylands, 2015-08-12:
The right place to put the print would be after line 25:
https://github.com/micropython/micropython/blob/master/py/makeversionhdr.py#L25 (before the if)

ha5di, 2015-08-12:
[root@beta unix]# make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CalledProcessError()
Traceback (most recent call last):
  File "../py/makeversionhdr.py", line 108, in <module>
    make_version_header(sys.argv[1])
  File "../py/makeversionhdr.py", line 74, in make_version_header
    info = get_version_info_from_git()
  File "../py/makeversionhdr.py", line 27, in get_version_info_from_git
    if er.args[0] == 128:
IndexError: tuple index out of range
make: *** [build/genhdr/mpversion.h] Error 1

[root@beta unix]#

robert-hh, 2015-08-12:
Same experience here. I'm not using git for Micropython. The version of makeversionhdr.py which came with 1.4.4-196 works fine. This monrning fix with 1.4.5-1 did not change the picture. 1.4.5 compiles well under Windows with MINGW, even w/o git.

ha5di, 2015-08-12:
Just to confirm, building a git clone works.

dpgeorge, 2015-08-12:
@ha5di what version of Python do you have?  (When you type "python" on the command prompt, what does the banner print?)

ha5di, 2015-08-12:
piCore on Raspberry: python: 2.7.10, python3: 3.4.3

Same result on CentOS7 where only 2.7.5 installed

robert-hh, 2015-08-12:
About versions:
With debian jessie and mint 17.1 it's python 2.7.6. But the same happens with python 3.4.3.
Printing er.args before line 26 in "../py/makeversionhdr.py" shows an empty tuple.
With Windows it's pathon 3.4.3, but no git installed, so it does not execute line 26.
Still, the build fails only if git is installed but no Micropython git repository exists.

dpgeorge, 2015-08-12:
The problem is that Py2.7's CalledProcessError behaves differently to Py3's.  Need to use er.returncode instead of er.args[0].

robert-hh, 2015-08-12:
Confirmed. replacing line 26 by:
        if er.returncode == 128:
works for both Python 2.7.6 and 3.4.0

ha5di, 2015-08-12:
Yes, now it builds fine with this change on CentOS7

dpgeorge, 2015-08-12:
Thanks for testing and feedback!  Now hopefully fixed in 94ef8879cd28e7531ab214f093e184d74490f639.

closed 2015-08-12 by dpgeorge