micropython/micropython #1426: A problem in "staccel.py"

jonathanZq, 2015-08-14:
There is a LIS302DL&LIS3DSH driver in python, in the STM32F4DISC.
And my board(with LIS3DSH) run this file, but it can not work.
The initial in this file,
"self.spi = SPI(1, SPI.MASTER, baudrate=328125, polarity=0, phase=1, bits=8)"
According to the datasheet, I think it should be "bits=16", and I had verified it.
but I do not know how to deal with LIS302DL, I do not have it.

dpgeorge, 2015-08-14:
Reading the datasheet for the LIS3DSH (Doc ID 022405 Rev 1) I see that SPI reads and writes are 2 or more bytes long.  I simple register read does take 16 clocks/bits, but that's implemented as 2 bytes: write the address of the register to read as the first byte, then read or write the data as the second byte.  So it should work with bits=8.  I'm quite sure that it has been verified to work with the LIS3DSH.  See commit 2b4af54992e1a4f2a3e02d5b72045cde60196e84.

pfalcon, 2015-09-13:
Assuming fixed, feel free to add more info to reopen.

closed 2015-09-13 by pfalcon

dpgeorge, 2015-09-15:
Probably the issue is the same as #1456 (SPI issue with DMA).  SPI DMA needs to be fixed, and probably staccel.py should be moved to drivers/ directory.

dpgeorge, 2015-09-15:
@jonathanZq this might now be fixed by d80174d7c3fe36f139c639d6a820cc4cbb88ddf9.

jonathanZq, 2015-09-16:
I am sorry, I have forgot to reply this.

I had read some source code, and I modified the "def read_bytes(self, addr, nbytes):", "self.spi.send" and "self.spi.recv" replaced by "self.spi.send_recv", then it can work correctly with bits=8.
And I did not fixed it according d80174d.
So, maybe other reasons?

Thanks for your reply!