micropython/micropython #1430: New MicroPython hardware API

danicampora, 2015-08-16:
Since we have the WiPy TODO list (#1425) Damien and I spent some time today discussing how to do things properly, once and for all, and do it before the WiPy ships. It involves a redefinition of the API in many aspects and it has implications all over the place. I'll do my best to layout all the points we touched:

- Use attributes instead of functions when we simply need to get/set things. I am going to quote @dhylands on issue #378.

> I think that to use attributes then you need to use nouns. If you can't represent it as a noun, then I think it should stay as a function

So for instance, `pin.value()` will change to `pin.value` to get it and `pin.value = x` to set it. This will be the way to go in other peripherals, e.g. timer.frequency, timer.period, etc.

- The whole `Pin` API will look like this:

```python
# create & init a pin
# id, mode are mandatory and positional (also can be named)
# pull is optional and positional (also can be named)
# rest of args are kwonly
# only "value" is required for a port to implement (initial value if given)
# rest are optional, and a port can define them and also define others
pin = Pin(id, mode, pull=Pin.PULL_NONE, *, value, drive, alt, slew, ...)

# constants
# OPEN_DRAIN is much more descriptive than OD
for mode: Pin.IN, Pin.OUT, Pin.OPEN_DRAIN, Pin.ALT, Pin.ALT_OPEN_DRAIN
for pull: Pin.PULL_UP, Pin.PULL_DOWN, Pin.PULL_NONE
optional pull: Pin.PULL_UP_STRONG, Pin.PULL_DOWN_WEAK, ...
for drive: Pin.LOW_POWER, Pin.MED_POWER, Pin.HIGH_POWER
for slew: Pin.FAST_RISE, Pin.SLOW_RISE

pin.init(...) # re init

pin.high() # set high
pin.low() # set low

pin.value # get value
pin.value = x # set value

# toggle
pin.toggle()

# attributes to get/set most common settings
# this duplicates a bit what init can already do, but it is useful in the case of pins
# for instance when bit banging like it's done by the one-wire driver,
pin.id # this one is read-only
pin.mode
pin.pull
pin.drive
...
```

- In general then any extensions to a function/method can happen in 3 ways:
    1. the port provides extra constants for an existing arg (eg Pin.PULL_UP_WEAK).
    2. the port provides extra keyword args.
    3. the port provides extra functions/methods.

- A new module called `hardware` will be created to replace the `pyb` module. `pyb` will still exist in the pyboard in order to provide backwards compatibility and to avoid confusions between the old and the new API.

- The new `hardware` module will contain `Pin`, `SPI`, `UART`, `I2C`, `Timer`, `CAN`,  and so on.

- The new hardware classes will use attributes where applicable.

- A new module named `machine` will hold all functions related to reset, interrupt and sleep management.

- The `UART` already supports the stream protocol and we would like to bring it to `SPI` and `I2C` as well. Right now we have the recv functions that can receive an integer or a buffer object, which is kind of weird and can get confusing. If we make `SPI` and `I2C` stream capable, then we have `read/readinto`, `write/writechar`. This will make the API across communication peripherals consistent and homogeneous. One problem of the stream approach is how to select the slave (for SPI) or the destination address (for I2C). We had two ideas:

    1. Simply add kwarg to read/write like `addr`, `memaddr`, `slave_addr`, etc. This will deviate from the standard stream API.

    2. Create and `endpoint` subclass which is the one actually providing the read/write methods and holds the information relevant to the destination like chip select, i2c address, addres size, etc. Doesn't solve the issue of having to add kwarg to the read/write methods in the case of the `memaddr` for I2C (not practical to have an endpoint for each memaddr).

I prefer option 1...

It's a lot to discuss in one ticket, but we think it's better because all things are related and it shows the complete picture.

The idea is to have this new API implemented (for the WiPy at least) in 2 weeks...

dpgeorge, 2015-08-16:
There are still some leftover functions/classes in pyb that don't yet have a home (eg LED, Servo, Accel, USB stuff).  But these are not present on the WiPy so deciding upon them is not urgent.  They will stay in pyb for now.  Most likely the USB stuff will move to a separate module, so that boards that have USB will have this module, and boards that don't, won't.

danicampora, 2015-08-16:
I think  that the whole `hardware` word already lets you know that what you see there relates to the hardware you are running MicroPython on. Your board doesn't have USB? then it doesn't show in hardware. You can do `dir(hardware)` and see what's supported. So, USB and SD (SDcard) which are very common but some boards have it and some not, should fit well into the `hardware` module. Maybe LED too...

Specific stuff like `Servo` and `Accel` can go into something like `pyboard`. The WiPy for instance has a `HeartBeat` class. This could go into `wipy`.

dhylands, 2015-08-16:
Things like UART and SPI are low level peripherals.

Things like Servo and LED are built on top of this. I'd like to see Servo and LED be much more flexible than they currently are. Currently it's impossible for the user to add additional Servos or LEDs to the ones which are "builtin". In theory I should be able to instantiate a Servo and associate it with a timer and/or a pin. For Servo and LED there are each a couple of different kinds.

1 - For Servo, you can have a Servo which is tied to a PWM output (much like the current pyb.Servo class). But you can also use a single timer to drive multiple servos which are connected to arbitrary GPIOs that don't necessarily have a PWM output (this is the model that teensy uses).

2 - Similarly, for LED, you can have a simple on/off LED, or you could have a PWM controlled LED.

I think that designing some of these "higher level" classes to be a bit more flexible makes them much more usable across various boards and board setups.

dpgeorge, 2015-08-17:
Regarding attributes vs functions (as previously discussed in #378, but this point I'm making here was not raised in that issue): the big downside of using attributes is that assignment is a statement and not an expression, so you would not be able to set the pin value (or do other attr assignments) in a lambda.  Eg you can't do:
```python
timer.callback(lambda tim: pin.value = 1)
```
With functions acting as getters/setters (based on whether 0 or 1 argument is supplied), which is how things work at the moment, you can do:
```python
timer.callback(lambda tim: pin.value(1))
```
I think this is a deal-breaker for using attributes.  Ignoring the implementation details and/or efficiency, and looking instead from a user/programmability point of view, the API should always use functions because they are expressions (more powerful than statements) and you can take a "pointer" to a function (eg pin_value_fun = pin.value) but not an attribute.

So, even though it's more Pythonic to use attributes for things that are clearly accessed using a getter and a setter (eg pin.value=1), I propose that we instead use functions that take 0 or 1 args (eg pin.value(1)).

danicampora, 2015-08-17:
> So, even though it's more Pythonic to use attributes for things that are clearly accessed using a getter and a setter (eg pin.value=1), I propose that we instead use functions that take 0 or 1 args (eg pin.value(1)).

OK, let's only use functions then.

pfalcon, 2015-08-17:
> the API should always use functions because 

I'm glad that we came to the same conclusion as we did before ;-).

dpgeorge, 2015-08-17:
> I'm glad that we came to the same conclusion as we did before ;-).

Yes indeed.  But since the topic here is to discuss a complete revamp of the API, I think it's useful that we had the discussion (again, but in a new light).

danicampora, 2015-08-17:
Just updated the original post scratching all the attribute ideas.

dpgeorge, 2015-08-17:
So do we then want to provide a "call" method as a faster way of getting/setting the pin:
```python
pin() # get pin value
pin(1) # set pin value
```
(this will be in addition to having the pin.value method)?

Rationale: pin getting/setting is a very common operation so benefits from a shortcut (shorter to type, smaller script, smaller bytecode).  Also, it's common to bit-bang using pins, and doing a call is much faster than looking up the method (pin() is roughly 3 times faster than pin.value()).

danicampora, 2015-08-17:
> Rationale: pin getting/setting is a very common operation so benefits from a shortcut (shorter to type, smaller script, smaller bytecode). Also, it's common to bit-bang using pins, and doing a call is much faster than looking up the method (pin() is roughly 3 times faster than pin.value()).

Even though it might seem a bit adhoc, and that it also duplicates what `pin.value()` already does, given the advantages, I think that in this case is a good idea and we should have the call method to set/get the value of a pin.

danicampora, 2015-08-18:
Right now there's a `pyb.mount()` function which works like this:

```python
pyb.mount(device, mount_point, *, readonly, mkfs)
```

I propose to move this function to the `os` module and remove the `mkfs` option. So it ends up being:

```python
os.mount(device, mount_point, *, readonly)
```

and then add a separate function, also in the `os` module, to make a file system, i.e.:

```python
os.mkfs(device or path, *, options...)
```

The required parameter can be a device just like the one accepted by `os.mount` or a path in case that the device is already mounted (like `/flash`). Maybe we need some keyword only arguments for extra option(s), maybe not.

Thoughts?

danicampora, 2015-08-20:
@dpgeorge asked me to create a wiki page to document the new API. I created it, and added the `Pin` class: https://github.com/micropython/micropython/wiki/Hardware-API

I've also implemented the new API for the WiPy. For now is in my personal repo:
https://github.com/danicampora/micropython/commit/0061f9f98fd9566a6fd6c560e60506321821fba8

pfalcon, 2015-08-20:
So, I saw https://github.com/micropython/micropython/wiki/Hardware-API (is that right place to do documentation? docs should be in RST form in repo/on a site, RFC is in tracker, not sure how overflow to wiki happens), and wonder: how did we end up with yet another "hardware" module? Last time it was discussed we seemed to agree that "machine" would be a place to transition elaborated APIs to (from "pyb").

danicampora, 2015-08-20:
> So, I saw https://github.com/micropython/micropython/wiki/Hardware-API (is that right place to do documentation? docs should be in RST form in repo/on a site, RFC is in tracker, not sure how overflow to wiki happens), 

OK, sorry, I corrected my post. The documentation itself will be done via sphinx just as before, the idea of the wiki is to show how the new API will look like before actually implementing it. It's easier to see it there than via the discussion in this PR. We can remove that link when the API is finished.

> and wonder: how did we end up with yet another "hardware" module? Last time it was discussed we seemed to agree that "machine" would be a place to transition elaborated APIs to (from "pyb").

Well, the new idea is to use `machine` for reset, interrupt, power management, and memory access.
`hardware` will contain the peripherals.

```python
machine.idle()
machine.sleep()
machine.enable_irq()
machine.reset()
machine.mem8()

hardware.SPI
hardware.UART
hardware.Timer
hardware.WDT
```

Don't you think it looks cleaner like that?

pfalcon, 2015-08-20:
> Don't you think it looks cleaner this way? ;-)

Not sure, I don't see clear pattern what should go into one module vs another. And having 2 modules instead of 1 takes more memory for boilerplate code.

dpgeorge, 2015-08-20:
Just to be clear, all of this API stuff is up for discussion.  Since there is a lot to discuss, and we want to make things as consistent and clean as possible across all peripherals, I thought it would be easier to have a single place where the current proposal is written down.  The github wiki seemed the best place (but could also be in this issue in the first comment but then it needs constant editing and will grow large).

The actual RFC discussion can happen right here.  @danicampora is implementing stuff before the API is finalised because he has hardware to ship very soon.

Regarding hardware vs machine module.  I thought one of the criticisms of pyb was that it was bloated with too many unrelated things.  Eg timing stuff should move to time module (as per #1225), generic "machine" stuff to machine module, etc.  In my mind, machine was about things a generic CPU core can do: access memory, reset, bootloader, sleep, interrupts, freq, etc.  And then hardware would be for more SoC specific additions, ie peripherals outside the CPU.  For pyboard's MCU, think of machine as abstracting what ARM provides, and hardware as what ST provides.  Then, as @dhylands suggests above, LED, servo and Accel are outside the SoC and in a different module again.

A blank module takes 11 words + qstr for the module name (around 60 bytes total for stmhal), which is not much, and I'd prefer to decide on a clean API without being too tied to minor things like this.

But again, this is all up for discussion, so please make your thoughts and alternatives known!

peterhinch, 2015-08-21:
This will surely break almost everything written for the Pyboard. Users will find a firmware update causes working code to fail. Contributed device drivers and utilities will fail unless the various authors promptly adapt their code. In the interim new users will find that the code on Github is broken. Is this wise?

dhylands, 2015-08-21:
I think that the plan is to also leave pyb for the pyboard, and eventually make it something that can be turned off.

The pyboard has enough flash memory to support both implementations.

danicampora, 2015-08-21:
> I think that the plan is to also leave pyb for the pyboard, and eventually make it something that can be turned off.

> The pyboard has enough flash memory to support both implementations.

Yes, that's exactly the plan.

peterhinch, 2015-08-21:
OK, thanks for that.

dpgeorge, 2015-08-21:
@peterhinch Just to reiterate: at the forefront of my mind is retaining compatibility with the existing pyb module and its API.  pyb will remain on pyboard and will remain the same.  The new modules (whatever they are called) will be in addition to pyb.  They will coexist and you can use pyb or the new stuff, or even a combination of both.  With updated firmware all existing code and drivers will still work.

The only drawback is that the WiPy won't be able to use existing code without some changes, but I think that's ok considering the benefits that the new modules/API will bring (consistent, clean, compatible across boards, simpler in some respects, and will allow to resolve lots of issues like DAC/ADC resolution).

peterhinch, 2015-08-22:
That all sounds good to me, and it does address some ongoing issues.

dpgeorge, 2015-08-22:
I think we should rename the I2C and SPI send/recv methods to write/read.  At the moment with UART using write/read you have to stop to think for a second if the current object you're using implements send/recv or write/read, and I think it's easier if everything uses write/read.  After all, transmission on I2C and SPI really is putting/getting a stream of bytes (not a packet like sockets) so write/read seems sensible.

I2C already has mem_read/mem_write.

To accomodate the fact that I2C needs to specify the address when in master mode we can use `readfrom(addr, len)` and `writeto(addr, buf)` (as well as `read(len)` and `write(buf)` for slave use).

danicampora, 2015-08-22:
I like it!

blmorris, 2015-08-22:
> I think we should rename the I2C and SPI send/recv methods to write/read. 

Sounds good to me. While trying to decide on names for I2S I looked at all the existing methods for a clear pattern- I didn't see one, so I settled on send/recv after SPI. I'm happy to change to write/read.

dpgeorge, 2015-08-22:
@blmorris thanks for feedback!  read/write are certainly good names for I2S (it really is a stream of data).

Also, we need to decide on numbering of peripherals, as used to create them in the constructor, eg UART(id, 9600, ...) -- what exactly is id?.  There are 2 options:

1. (current pyboard scheme) Number by manufacturer's datasheet.  STM32F has periphs starting at 1, and some are not sequential (eg no UART5 on pyboard, but does have a UART6), so you have UART 1, 2, 3, 4, 6.  On WiPy they start at 0, so you have UART 0, 1.  Pros: matches datasheet so you can lookup the periph easily to see its features (useful for Timer which are not all the same).  Cons: not Pythonic, can't iterate through them, not consistent acroos boards.

2. Number from 0, sequentially.  So pyboard UART would be 0, 1, 2, 3, 4 (mapping to hardware ids 1, 2, 3, 4, 6).  Pros: Pythonic, easy to index and iterate over (eg `[UART(id) for id in range(5)]`), consistency across boards (so `UART(0)` will always exist).  Cons: need to provide a table in the docs matching this 0-based sequential id to hardware id.

dhylands, 2015-08-22:
I think that you could still iterate over them, just not by index. You'd instead do something like
```
for uart in pyb.UART.all():
```
which would construct a tuple (or something) of all of the existing UARTs. If we made the UARTs go from 0-N then i think we'd need to add an address or something which corresponds to the peripheral ID (i.e. know that it's UART 6 on the pyboard).

dpgeorge, 2015-08-22:
> `for uart in pyb.UART.all()`:

I thought about this, but then UART.all() would need to actually construct all the uart instances (eg allocate memory for their objects), which is not really what you want to do.  And you (arguably) want to be able to iterate over all uarts, not just those that have been already constructed.

Instead UART.all could be a tuple of all the ids, eg: `UART.all -> (1, 2, 3, 4, 6)`.

blmorris, 2015-08-22:
I2S2 & I2S3 is certainly not pythonic, but giving I2S3 an index of either 0 or 1 depending on whether I2S2 is enabled prevents one from knowing exactly what hardware is being pointed to.
I imagine a truly cross-board compatible script will need to have a way to know what board it is running on anyway.

chrismas9, 2015-08-23:
Also, we need to decide on numbering of peripherals

-1 for MCU peripheral numbers. For code portability I would prefer names that are related to the pinout of the module family. We already have this for GPIO with the X and Y numbers. ST make a wide range of nominally pin compatible devices suitable for uPy including the L152 and F446 (lower power / faster speed) but the peripheral multiplexing is different. PA0 is TIM5ch1 on F405 but TIM2ch1 on F411. There are a wide range of pin compatible Discovery boards but their alternate function mapping to board pins won't be consistent.

    STM32F has periphs starting at 1

Not always. On some STM32 the ADC's start at ch0 (ef F405), on others they start at ch1 (eg F303). Some have DAC1 and DAC2, others have DAC1ch1 and DAC1ch2, etc

My preferred options are:

1. Sequential, eg UART1, UART2, UART3, etc. or

2. Based on board pin name. For multi pin ports use the first output pin (Tx, CLK, etc). This yields GPIO(X1), PWM(X1), UART(X1), UART(X3), SPI(X6), SPI(Y6), I2C(X9), I2C(Y9).

We already have this board pin name convention for single pin functions - GPIO(X1), PWM(X1), ADC(X1), DAC(X1). Why not use the same convention for multipin peripherals where there is a pin allocation already assigned for the board.

MicroPython is very good at insulating users from the peculiarity and complications of the underlying hardware. Option 2 is easiest for most users. Wire a GPS to pins X1 and X2. Talk to it with UART(X1). It also gives consistent naming for all alternate functions on the same pin, eg X1 can be GPIO(X1), PWM(X1), ADC(X1) or UART(X1).

For the few cases where it is necessary to know the underlying peripheral device this would need to documented.

dhylands, 2015-08-23:
In general, I think that this is a good idea. However, we need to deal with the case where multiple instances of the same peripheral exist on the same pins.

For example, on the '405
PA15 has SPI1_NSS and SPI3_NSS
PB3 has SPI1_SCK and SPI3_SCK
PB4 has SPI1_MISO and SPI3_MISO
PB5 has SPI1_MOSI and SPI3_MOSI

For some boards this would be resolved at the board level, but for more generic boards, like the discovery boards, which one to chose becomes a function of how the board is being used.

The timers also do this. You can look at http://micropython.org/resources/pybv10-pinout.jpg and see that in some cases, up to 3 timers are mapped to the same pin.

chrismas9, 2015-08-23:
However, we need to deal with the case where multiple instances of the same peripheral exist on the same pins. For some boards this would be resolved at the board level, but for more generic boards, like the discovery boards, which one to chose becomes a function of how the board is being used.

For boards like Pyboard the peripheral pins are already assigned. Maybe for generic boards users could create a pin map to assign a set of peripherals to chosen pins. Once created for a board scripts would then be compatible independent of the underlying hardware and MCU.

I thought about using sequential numbers or sequential names such s SPI1, SPI2 assigned to board pins. The problem is when a new MCU comes out with more peripherals you have to add more numbers and they won't be in order around the board. Also for industry standard pinout such as Arduino which is used on many boards such as Nucleo, Freedom, etc different people might assign sequential numbers to different pins and that breaks code portability too.

For timers and PWM, etc I would like to see a default peripheral assigned to pins so users can just set a period or duty cycle for simple applications, but be able to override the default when they need more control or choice, eg to use different frequencies for PWM on different pins.

chrismas9, 2015-08-23:
but for more generic boards, like the discovery boards, which one to chose becomes a function of how the board is being used.

For new ports using machine and the new API would it be reasonable to have a default peripheral to pin map with a coloured pin chart for each supported platform. There's nothing to stop people using the MCU port name (PA0, etc) instead of the board pin name (X1, etc) if they want a different allocation. Pyboard already has a standard mapping. Wipy has two UART0_TX pins, it would just be a matter of nominating one of them as the default on the IO chart.

danicampora, 2015-08-23:
Here's my take on the peripheral IDs:

Make them sequential and starting at 0 (0,1,2,3,4, ..). This is the most Pythonic and since we have datasheets that start peripheral IDs at 0 and others at 1, we should simply follow the Pythonic way. I think than whenever in doubt, making the API more in line with what people is used to do in Python should be the way to go.

All peripherals will have the same methods that will work the same way, but the constructor needs info about the `ID` and the `Pins`, and this is totally different for each board. There are two options as I see it:

1. The library receives an instance of the peripheral.
2. The library receives the `ID` and the pins of the peripheral.

In any case, what the IDs and the Pins are, is irrelevant for code portability (at library level, a complete script won't be portable between tow different boards). 

A small reference chart indicating which Python IDs relate to which datasheet IDs is enough.

danicampora, 2015-08-23:
> For new ports using machine and the new API would it be reasonable to have a default peripheral to pin map with a coloured pin chart for each supported platform.

Yes, that's what we should do as well. Is not pins passed to a constructor, it takes the default for that peripheral ID, and those defaults must be documented in the reference chart. The WiPy has this:

https://github.com/wipy/wipy/blob/master/docs/PinOUT.pdf

And it will be updated and completed once our new API is ready.

chrismas9, 2015-08-23:
Make them sequential and starting at 0 (0,1,2,3,4, ..). 

This works well for a single board but doesn't solve the problem for a family based on a common board format and pin allocation. If you start with an STM32F405 with 5 uarts U0 .. U4 assigned to particular board pins then do an F401 board with 3 uarts how do you number them _U0 .. U2? What if U2 is on the same pins as U5 on the original board. Or does the F401 board have U0, U1 and U5 for compatibility with the F405 board? This is no longer sequential.

I would rather have the same name, eg UART(X1) across all boards in a family irrespective of MCU.  

For the new API what is proposed for single pin functions like ADC which use the board pin name now? Will they still use board pin name or be sequential numbers (ADC0 .. ADC10)?

danicampora, 2015-08-23:
> This works well for a single board but doesn't solve the problem for a family based on a common board format and pin allocation. If you start with an STM32F405 with 5 uarts U0 .. U4 assigned to [...]

First of all, we shouldn't try write the API based on what suits best a single family of MCUs (STM32F4 in your example). 

When looking at the bigger picture, UART will be named in the datasheet in many different ways, and pins will also be named different. In my opinion, the goal should be to have an API that is consistent and works exactly the same no matter if its an ST, TI, Atmel or whatever brand MCU. In general you know that if it has an UART, there's at least UART0, and if there are more, they will follow go from 0 to n-1. You want to know to which pins UART0 is connected by default? just take a look at the reference chart of the board instead of having to read the 1000s pages datasheet. That reference sheet MUST also mention which timers are 32bit, which are 16 bit, and any other important detail. I am pretty sure that we can achieve that easily for almost any MCU that we aim to support.

dhylands, 2015-08-23:
ok - what if we did something like the following:

1 - When you initialize a UART, you can specify an ID or a Pin. If, by chance, there are multiple peripherals available on the pin, the board will designate one of them as a default.
2 - I'd also like to be able to pass individual pins, so for example, a UART can have TX/RX/CTS/RTS. So yo might do something where you specify u = UART(3, rx_pin='X1', tx_pin='X2', rts_pin='X4', ...)

I'm still not sure about the IDs. I'd be quite happy to say that the IDs can start at 0 and increment by 1 as long as we also had a way to specify IDs by the cpu identifier.

If we allow the pins to used as an id, then perhaps we could also allow a string like 'cpu.5'.
Or perhaps we could have pyb.UART.board.UART0 and pyb.UART.cpu.UART0 (much like the pins)

danicampora, 2015-08-23:
> 2 - I'd also like to be able to pass individual pins, so for example, a UART can have TX/RX/CTS/RTS. So yo might do something where you specify u = UART(3, rx_pin='X1', tx_pin='X2', rts_pin='X4', ...)

@dhylands what you suggest has been proposed in the new wiki section describing the API, please take a look in:
https://github.com/micropython/micropython/wiki/Hardware-API#the-uart-class

The idea is to use this same (or similar) approach for other peripherals. If you don't specify pins, then the default ones for that peripheral are used.

> I'm still not sure about the IDs. I'd be quite happy to say that the IDs can start at 0 and increment by 1 as long as we also had a way to specify IDs by the cpu identifier.

I don't know if it's wise to provide several options to specify the ID, there should only be one obvious way to do it.

dhylands, 2015-08-23:
My only comment about the pins (on the hardware wiki) is that I'd prefer to see kwars used rather than an array. Using an array, there is nothing "self describing". I'd much rather see rx_pin=X1, tx_pin=X2 rather than pins=(X1, X2). I think using kwargs also simplifies the underlying C code.

danicampora, 2015-08-23:
> My only comment about the pins (on the hardware wiki) is that I'd prefer to see kwargs used rather than an array. Using an array, there is nothing "self describing". I'd much rather see rx_pin=X1, tx_pin=X2 rather than pins=(X1, X2). I think using kwargs also simplifies the underlying C code.

OK, that seems sensible.

dpgeorge, 2015-08-23:
> My only comment about the pins (on the hardware wiki) is that I'd prefer to see kwars used rather than an array. Using an array, there is nothing "self describing". I'd much rather see rx_pin=X1, tx_pin=X2 rather than pins=(X1, X2).

I agree that it's much more self documenting to have kwargs, one for each pin.  But it really does blow out the constructor definition (uart's going to have >10 parameters).  We also want to support RTS and CTS.  And SPI needs 3 or 4 pins specified.  Eg for uart you'd have:
```
UART(1, 9600, 8, None, 1, pins=(Pin('GP1'), Pin('GP2')), timeout_char=10, read_buf_len=64)
# versus
UART(1, 9600, 8, None, 1, tx_pin=Pin('GP1'), rx_pin=Pin('GP2'), timeout_char=10, read_buf_len=64)
```
The first form (pins) seems simpler and is then consistent across peripherals (they all use the pins argument).  The second is self documenting but a bit longer to write.

> I think using kwargs also simplifies the underlying C code.

This is not a good reason to decide how the API will look.  The API should favour the end user, not the implementor.

danicampora, 2015-08-23:
> This is not a good reason to decide how the API will look. The API should favour the end user, not the implementor.

This is true. I also think that it will make the constructor definition too long.

One thing, besides `pins=(Pin('GP1'), Pin('GP2'))` with pin objects, I think it should also accept pin IDs directly, just like the `Pin` constructor itself. So that you can also do:

```python
UART(1, 9600, 8, None, 1, pins=('GP1', 'GP2'), timeout_char=10, read_buf_len=64)
```

blmorris, 2015-08-23:
> I'd also like to be able to pass individual pins, so for example, a UART can have TX/RX/CTS/RTS. So yo might do something where you specify u = UART(3, rx_pin='X1', tx_pin='X2', rts_pin='X4', ...)

+1 on this from me. I have this implemented this for I2S for the simple reason that the pins used for I2S on my board don't match the SPI2 pins on the pyboard. 
My current implementation takes a list of pin descriptors (could be board strings like 'X7', cpu strings like 'B12', or initialized pin objects. It verifies that they describe a valid I2S port and returns an I2S object. Note that you don't pass an ID in this scheme, the instance is specified only by the pins.
Currently my code takes a list of pins; that seemed simplest to me at the time. I would agree that it isn't self-describing; kw_args may be better.

danicampora, 2015-08-23:
> +1 on this from me. I have this implemented this for I2S for the simple reason that the pins used for I2S on my board don't match the SPI2 pins on the pyboard.

Yes, a list of pins is going to be part of the API for constructors for sure. 

> Note that you don't pass an ID in this scheme, the instance is specified only by the pins.

The problem with this is that for some MCUs a given pin might be multiplexed by both UART0 and UART1 for instance. This is why we propose to pass the ID as well.

blmorris, 2015-08-23:
My pin-oriented initialization code is at https://github.com/blmorris/micropython/blob/967f020819b8d2b7a84ac96b63b3327b182fa6d6/stmhal/i2s.c

See 'pyb_i2s_make_new'
(Writing from my phone, can't give a link to the exact lines of code).

blmorris, 2015-08-23:
> The problem with this is that for some MCUs a given pin might be multiplexed by both UART0 and UART1 for instance. This is why we propose to pass the ID as well.

That is true, but I don't think a full list of pins will ever be a match across two instances. I could be wrong, and either way it is easy enough to just require an instance ID anyway. It would be redundant, but it would also simplify the validation code.

danicampora, 2015-08-23:
@blmorris we also want to allow stuff like:

```python
uart = UART(0, 9600)
```

A quick initialization which takes the default set of pins for UART0.

blmorris, 2015-08-23:
@danicampora -
> @blmorris we also want to allow stuff like:

>>uart = UART(0, 9600)

>A quick initialization which takes the default set of pins for UART0.

Point taken. I allowed for a default set of pins when no pin args were passed, but I won't claim that my way was the correct way to do it. I'll follow your lead here; I'm glad the API will offer a standard way to do what I need.

hoihu, 2015-08-25:
why not something like:
```python
> import UART
> uart = UART(10, 9600)
ImportError: UART10 not supported on this board

> # so where are my uarts?
> print UART.list_ports()
[ 
{'id': 0, 'doc':'UART1' , 'mapping': {'rx_pin': ['X1','X2','X3'], 'tx_pin':['X4','X5','X6'], 'rts_pin':['Y1'], 'cts_pin':['Y2'] }
{'id': 1, 'doc':'UART2' , 'mapping': {'rx_pin': ['X1','X2','X3'], 'tx_pin':['X4','X5','X6'] }
]

> # use UART2
> uart2 = UART(1,9600, rxpin = 'X1', txpin = 'X4')
> # or maybe if we just want to send something and not wasting an RX pin 
> # not sure if that's already possible in the existing uPy API..
> uart2 = UART(1,9600, txpin = 'X4')
>
> # further - checks would be nice like
> uart2 = UART(1,9600, rxpin = 'X5', txpin = 'X4')
ImportError: UART rx pin on X5 not supported
```
The mapping could be specified in a JSON file (machine.json?) per board and compiled as a *.h file at build time - similar to the existing CSV file. 

Alternatively, it might be possible to move the "mapping" section to the pin class and provide there  "capabilities" - so the pin class can accept/receict if the UART wants to use it as UARTTX for example. 

Regarding the wiki example:

```python
UART(1, 9600, pins=('GP1', 'GP2')) 
# enable the pull-ups on both UART1 pins
Pin('GP1', mode=Pin.ALT, pull=Pin.PULL_UP)
```
I'd propose not to encourage people to modify Pin inits after the module is initialized, as this might lead to not working UARTs, I2Cs buses.

dhylands, 2015-08-25:
I think that most of the information to generate the mapping already exists in the pins files, just not grouped by peripheral.

We generate a pins_af.py file which is placed in the build directory and this is a python tuple which lists all of the alternate functions for each pin contained in pins.csv.

It should be easy to walk this and produce a similar data structure which lists the pins per peripheral (and I think that would be a useful file to generate as well).

danicampora, 2015-08-25:
> I'd propose not to encourage people to modify Pin inits after the module is initialized, as this might lead to not working UARTs, I2Cs buses.

The idea is not to encourage people to do that, but to show that the API is flexible enough to allow such "advanced" settings, which are certainly useful if you know what you are doing.

hoihu, 2015-08-25:
> The idea is not to encourage people to do that,..

encourage was the wrong word I guess. I just couldn't think of a use-case. didn't want to offend.
> I think that most of the information to generate the mapping already exists in the pins files, just not grouped by peripheral.

My idea was to extend stm32fxxx_af.csv with additional informations that would allow the mapping to be possible. For this a more powerful data structure than CSV might be needed (and JSON is quite popular these days).

Something like this (in stm32fxxx.json)
```json
{
  "Pins": [
        { "id": "X1", "cpu_port": "PA", "cpu_pin": 0,
          "af": [
                { "af_nr": 0 , "from": "USART1_RX"}
                { "af_nr": 3 , "from": "USART1_TX"}
                { "af_nr": 10 ,"from": "SPI1_MISO"}
          ],
          "drive_strengths" : ["low", "high"],
          "pullup" : ["No", "yes"]
        }, 
        { "id": "X2", "cpu_port": "PA", "cpu_pin": 1,
          "af": [
                { "af_nr": 3 , "from": "USART2_RX"}
                { "af_nr": 5 , "from": "USART2_TX"}
                { "af_nr": 9 ,"from":  "SPI2_MISO"}
          ],
          "drive_strengths" : ["low", "high"],
          "pullup" : ["No", "yes"]
        },
        ...
        
  "uarts": [
        { "id": "0", "cpu_name": "USART1", 
           "wires" : [
                {'rx': "USART1_RX"}, 
                {'tx': "USART1_TX"},
                {'rts': "RTS1"},
                {'cts': "CTS1"}
            ],
        },  
    ]
    
    "spi": ...
    "i2c": ...
}  
```
That's just an example and I'm not happy with the drive_strengths and pullups. 

But in such a way, a mapping from USART1 to the pins is obvious and can be checked at build time ("wires" and "from" must match). It can then be processed by a python script to generate the required help information for "list_ports"

I know that it would require a lot of work to do it but it would allow to centralize the HW stuff in one JSON file that can be extended easily with future entries

dhylands, 2015-08-26:
Like I said, I don't think that you need to extend anything. All of the information you're looking for is already present in stm32f4xx_af.csv, it just needs to be pulled out.

And the real secret is that stm32f4xx_af.csv defines the alternate functions for all 4xx chips. pins.csv defines which pins are valid for a given board, so we more or less do an intersection of the 2 files to determine which alternate functions are available for a given board.

roger-, 2015-08-26:
> I know that it would require a lot of work to do it but it would allow to centralize the HW stuff in one JSON file that can be extended easily with future entries

Sounds a bit like a Linux device tree file (DTS).

danicampora, 2015-08-26:
> encourage was the wrong word I guess. I just couldn't think of a use-case. didn't want to offend.

No offense taken ;-).

blmorris, 2015-08-27:
I am assuming that my I2S code will get merged once it is ready, so if there are no objections I will add an I2S section to the Hardware API wiki page. The new API concepts will require me to make some adjustments anyway; having an I2S section on the wiki will help to establish a fixed target as I get the code ready to merge.

danicampora, 2015-08-27:
@blmorris sounds good! go ahead :-)

hardkrash, 2015-08-30:
From my experience with USB, UART, GPIB, I2C, and SPI interfaces in Python making read and write take an extra keyword parameter is not an easy to maintain or extend mechanism.  I've rewritten an instrument automation stack 4 times and the first three times I did option "i" cause it was "easier" the last rewrite was with buses and and endpoints.  It has survived and been extended by many now.

If a library that takes to a device just works with a stream with read write then you going to work well with UART but badly with I2C and SPI with those addresses and slave select keywords.  You now have to wrap them to get the functionality.  The users are going to wrap them oddly or use the boilerplate you're going to write on the wiki.

I have implemented the option "ii" with the endpoint that references the bus for communications.  On my PC  the BUS even has a lock to prevent intermixed communications.

You always have to wrap the interface when using libraries, let's cleanly wrap them with the hardware API and provide the stream based endpoints as a streamed endpoint class.  If we have concepts of a mutex available we should implement them.  I once agreed that an endpoint wrapper was easy to write, but my fellow non-programmers got easily confused with the boiler plate.

This also enables:
- A clean interface for RS-485 devices.
- Method for muxing UARTs with hardware. (Think slave select applied to UARTS)
- Ability to have a self contained muxed I2C bus.
    - ```dev_ep = hardware.I2CEndpoint(nxp.PCA9547(hardware.I2C(opts), port), dev_addr=0x0D)```

```python
import dev_lib
my_i2c_bus = hardware.I2C(opts)
my_i2c_device = my_i2c_bus.endpoint(0x10)  # Call class factory to get endpoint for address 0x10
    # In my implementation I store a weak reference to the endpoint class.  e.g. bus.register_endpoint(self)
    # Store who currently has the mutex.
    # could be a one liner if you only had one I2C device.
        # my_i2c_device = hardware.I2C(opts).endpoint(0x10)

my_dev = dev_lib.fancy_dev(mode='i2c', interface=my_i2c_device)
```

hardkrash, 2015-08-30:
Please change watchdog.kick() to watchdog.feed()
I suspect many children will be using this board.

I rather not kick dogs...  anyhow a cartoon burger often gets out a T-Bone steak for feed the watchdog and prevent the alarm from sounding.

[270 more replies]

closed 2019-05-21 by dpgeorge