micropython/micropython #1435: Single precision float printing error

dpgeorge, 2015-08-20:
When uPy is build with single-precision float (unix or stmhal port) some numbers close to 0.1 are printed as 1.1.  Eg:
```python
>>> from array import array
>>> array('f', b'\xcc\xcc\xcc=') # representation of 0.1
array('f', [1.1])
```
It only fails if the first byte in the representation is between 0xc6 and 0xcc inclusive.  There are probably other numbers that also don't print correctly, and double precision may be broken as well (but no proof of this).

ha5di, 2015-08-27:
Running on piCore-7.0alpha2 (Raspberri Pi 2):

# micropython
Micro Python v1.4.5-27-g5ab0a4a on 2015-08-25; linux version
>>> from array import array
>>> array('f', b'\xcc\xcc\xcc=')
array('f', [0.09999999403953552])
>>>

pfalcon, 2015-08-27:
@ha5di : Did you build piCore version with single-precision floats? (If so, why?)

pfalcon, 2015-08-27:
I changed priority from low to high - not printing correct numbers is IMHO rather big problem. @dhylands : do you plan to look into this?

ha5di, 2015-08-27:
Sorry, I must be wrong. I changed nothing in the config, used default setting. Now I guess single precision is not the default :(

pfalcon, 2015-08-27:
Yes, the topic of this bug is that uPy has own routine for single-precision floats, and it has bug. But single-precision floats are useful only for hardware with SFP coprocessor or w/o coprocessor at all, as optimization. Otherwise, doubles are used by default, and they're printed by libc.

ha5di, 2015-08-27:
Sorry for making unneded noise :(

pfalcon, 2015-08-27:
@ha5di : Your feedback is appreciated, no problem, please keep it coming. And we'd hope that you keep contributing, so happy to give hints about the internals to make this learning more smooth.

dhylands, 2015-08-27:
@pfalcon Yes - I'm planning on taking a look at it real soon. Been a bunch of family issues to deal with....

pfalcon, 2015-08-27:
@dhylands: Thanks, no hurry of course.

commit 9d6128ac (2015-09-11): stmhal: fix single precision float printing error

Fixes #1435.
closed 2015-09-11 by pfalcon