micropython/micropython #1436: Should we add an initcall mechanism?

dhylands, 2015-08-22:
Currently, if we code python types in C, we need to stick initialization function calls in main.

This proposal was triggered by http://forum.micropython.org/posting.php?mode=reply&f=3&t=871#pr5001

I'd like to propse a mechanism, essentially the same as what the linux kernel calls initcall. Basically, we'd create various categories of initialization calls, something along the lines of:

1 - Early init once (only ever call once)
2 - Early init (called before heap initialization, but called for 1st init and after soft reset)
3 - INIT0 - called after the heap is initialized but before boot.py
4 - INIT1 - called after boot.py is initialize

The C code would look something like this:
```C
STATIC void my_init(void) {
    ...so some initilization...
}
MP_INIT1(my_init)
```

Basically MP_INIT1 just creates a function pointer which is saved away in a named section (say .init.0 instead of .text) and then main.c would do call all of the init1 functions by calling all of the function pointers in the .init.1 section at the appropriate time.

This would allow the C modules to decoupled from main.c

I know this is essentially trivial to implement in gcc. I'm pretty sure that most other compilers have similar mechanisms, but I can't swear to it.

pfalcon, 2015-08-22:
See https://github.com/micropython/micropython/commit/1f28413e840b6f688dbc9deda93ff4e33984019e . For builtin modules, it hardly makes sense to go long way trying to "decouple" them from main.c.

dhylands, 2015-08-22:
Cool - not sure why/how I missed that - but I'll definitely check it out in more detail.

dpgeorge, 2015-08-22:
That commit is in an unmerged branch (loadable-native).

Isn't the question on the forum about initialising a built-in module (like ujson) upon import (eg import ujson)?  If the module were written in Python then it would execute the script to import it and the script can do stuff, so the question is, is there a way to run some C function when a C module is imported.

dhylands, 2015-08-22:
Yeah - I think I got off track tthinking about the initialization stuff. 

With frozen modules, you could create a python module which references the C module and get the intiialization when the python module was imported (something that would work today with no changes).

pfalcon, 2015-09-13:
Assuming can be closed for now.

closed 2015-09-13 by pfalcon

dpgeorge, 2015-09-15:
I actually needed this feature myself (to enable implementation of "import this" natively, to print the Zen of (Micro)Python upon import), and had the following patch for it:

```diff
diff --git a/py/objmodule.c b/py/objmodule.c
index e0fad37..fac93b1 100644
--- a/py/objmodule.c
+++ b/py/objmodule.c
@@ -212,6 +212,15 @@ mp_obj_t mp_module_get(qstr module_name) {
         if (el == NULL) {
             return MP_OBJ_NULL;
         }
+
+        // look for __init__ and call it if it exists
+        mp_obj_t dest[2];
+        mp_load_method_maybe(el->value, MP_QSTR___init__, dest);
+        if (dest[0] != MP_OBJ_NULL) {
+            mp_call_method_n_kw(0, 0, dest);
+            // register module so __init__ is not called again
+            mp_module_register(module_name, el->value);
+        }
     }
 
     // module found, return it
```

If a native module defines `__init__` then this is called at time of first import.  Should we add such a feature?

danicampora, 2015-09-15:
@dpgeorge Sounds good to me. +1.

mianos, 2015-10-08:
I just patched this into my fork and it works perfectly. This is exactly what I was asking for on the forum. +1 to add it.

dpgeorge, 2015-10-08:
Ok, we can add it to master with an option like `MICROPY_MODULE_INIT`, disabled by default.

dpgeorge, 2015-10-12:
Now added in 3c9c3687d6c1001b88ef6dde200566456a1f2641.  Enable with `MICROPY_MODULE_BUILTIN_INIT`.  Disabled by default and currently disabled on all ports.