micropython/micropython #1440: How do I determine what version of MicroPython is on my Pyboard?  How do I upgrade to v1.4.5?

cclauss, 2015-08-27:
I can not find answers in the docs...
* How do I determine what version of MicroPython is on my Pyboard?
* How do I upgrade to v1.4.5? https://github.com/micropython/micropython/releases/tag/v1.4.5

dhylands, 2015-08-27:
The version is printed in the banner at the top when micropython starts, or when you press Control-D (soft reset). Something like:
```
Micro Python v1.4.4-180-g44e3df4 on 2015-08-06; PYBv1.0 with STM32F405RG
```

You can get new firmware versions from this page:
http://micropython.org/download/
There are links to pages on how to actually perforn the flashing on that page as well.

closed 2015-08-28 by cclauss

WizGeek, 2022-08-28:
For EV3, Open an SSH Terminal and issue:
`micropython -c 'import sys ; print("PyLang", sys.version_info, ";", sys.implementation)'`