micropython/micropython #1443: There is no way to query usb_mode

dhylands, 2015-08-29:
It would be useful (from a debugging perspective) to be able to query the current usb_mode.

I was thinking that calling pyb.usb_mode() with no arguments should return the currently configured usb mode.

dpgeorge, 2015-08-29:
Yes, returning the current mode by calling pyb.usb_mode() with no args seems sensible and consistent with other methods.

TDAbboud, 2015-08-30:
If we add this query, should we also alter the disabling of usb, which is currently, pyb.usb_mode(None). I could see someone disabling the usb by accident when they were trying to poll the current mode.

dhylands, 2015-08-30:
I was thinking about that. If my understanding is correct, calling pyb.usb_mode is only valid in boot.py prior to usb being initialized.

Looking at the source code, calling pyb.usb_mode(None) is a do-nothing function.

pfalcon, 2015-09-13:
Should be fixed by the patch above.

closed 2015-09-13 by pfalcon