micropython/micropython #1445: REPL tests are not reliable

pfalcon, 2015-08-30:
Here's result of random failure running with native unix version:

~~~~
4 tests skipped: repl_emacs_check repl_emacs_keys machine_mem extra_coverage
1 tests failed: repl_basic
~~~~

These tests always well when run with qemu user mode emulation for other archs (extra '\r' from a quick look).

dpgeorge, 2015-08-30:
Does repl_cont pass?  Does repl_basic always fail if you run it multiple times?  It uses up-arrow to test history so maybe there's something funny with the escape code processing.

pfalcon, 2015-09-01:
Here's random failure of repl_basic:

~~~~
--- repl_basic.py.exp	2015-09-01 11:44:30.439089919 +0300
+++ repl_basic.py.out	2015-09-01 11:44:30.439089919 +0300
@@ -1,5 +1,5 @@
-Micro Python \.\+ version
->>> # basic REPL tests
+# basic REPL tests
+Micro Python v1.4.4-262-g46e6a0c-dirty on 2015-09-01; linux version
 >>> print(1)
 1
 >>> print(1)
~~~~

dpgeorge, 2015-09-01:
Looks like an issue with system printf vs custom mp_printf, so order of output is messed up.

pfalcon, 2015-10-06:
https://travis-ci.org/pfalcon/micropython/builds/83911374

~~~~
1 tests failed: repl_cont
The command "(cd tests && MICROPY_CPYTHON3=python3.4 MICROPY_MICROPYTHON=../unix/micropython_coverage ./run-tests)" exited with 1.
~~~~

The same commit pushed to master worked ok.

pfalcon, 2015-10-06:
Well, I restarted that build without thinking, so the link above no longer shows that failure, but the point is: it happens, and even outside my local system.

pfalcon, 2015-11-24:
Fixed by https://github.com/micropython/micropython/commit/ff8c4188f473ac7c8311694a6ffa4ca5357b0492

closed 2015-11-24 by pfalcon