micropython/micropython #1448: Arithmetic comparison errors involving sys.maxsize

pramasoul, 2015-08-31:
```python
Micro Python v1.4.4-251-g21ffa7c on 2015-08-30; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import sys
>>> sys.maxsize == sys.maxsize - 1 + 1 # expect True                                                           
False
>>> sys.maxsize > sys.maxsize - 1 + 1  # expect False                                                          
True
```
The error occurs within ```sys.maxsize``` and ```2 * sys.maxsize + 1```:
```python
>>> sys.maxsize - 1 == sys.maxsize - 2 + 1 # expect True                                                       
True
>>> 2 * sys.maxsize + 1 == 2 * sys.maxsize - 1 + 1 + 1 # expect True                                           
False
>>> 2 * sys.maxsize + 2 == 2 * sys.maxsize - 1 + 1 + 2 # expect True                                           
True
```
Yet everywhere equivalence is expected, the string representations are equal:
```python
>>> str(sys.maxsize) == str(sys.maxsize - 1 + 1) # expect True                                                 
True
>>> str(2 * sys.maxsize + 1) == str(2 * sys.maxsize - 1 + 1 + 1) # expect True
True
```
The unix build behaves the same.

Note that both ```stmhal/mpconfigport.h``` and ```unix/mpconfigport.h``` have:
```C
#define MICROPY_LONGINT_IMPL        (MICROPY_LONGINT_IMPL_MPZ)
...
#define MICROPY_PY_SYS_MAXSIZE      (1)
```
which puts us in ```objint_mpz.c```. I had no luck finding odd behavior around bit crossings using:
```python
def test(n):
    x = 1
    for i in range(n):
        x += x+1
        if x == sys.maxsize:
            print("arithmetic == at", i+1, x, hex(x))
        if hex(x) == hex(sys.maxsize):
            print("hex() == at", i+1, x, hex(x))
        yield x-1 == x-1 - 1 + 1
        yield x   == x   - 1 + 1
        yield x+1 == x+1 - 1 + 1
```
On pyboard:
```python
>>> all(test(2050))
hex() == at 30 2147483647 0x7fffffff
True
```
And unix:
```python
>>> all(test(2050))
hex() == at 62 9223372036854775807 0x7fffffffffffffff
True
```
Cpython of course gives:
```python
>>> all(test(2050))
arithmetic == at 62 9223372036854775807 0x7fffffffffffffff
hex() == at 62 9223372036854775807 0x7fffffffffffffff
True
```

dpgeorge, 2015-08-31:
Nice find!  The problem is in objint_mpz.c, in the definition of mp_maxsize_obj.  I think the .len field is wrong (line 61).  Try making .len=2 on 64-bit unix.

pramasoul, 2015-08-31:
Making .len=2 on 64-bit unix does indeed give correct behavior. What preprocessor expression should I use in place of ```MPZ_NUM_DIG_FOR_INT``` to give the correct value across platforms? 

```C
const mp_obj_int_t mp_maxsize_obj = {
    {&mp_type_int},
    {.fixed_dig = 1, .len = MPZ_NUM_DIG_FOR_INT, .alloc = MPZ_NUM_DIG_FOR_INT, .dig = (mpz_dig_t*)maxsize_dig}                                                                                               
};
```

dpgeorge, 2015-09-01:
> What preprocessor expression should I use

I don't know, it has to be worked out.  Maybe the definition of MPZ_NUM_DIG_FOR_INT can be fixed.

alex-march, 2015-09-15:
I think the problem with the MPZ_NUM_DIG_FOR_INT macro was the +1 in it's definition. Removing it makes the number of digits stay at 2 for 32/64 bit unix. Let me know if it's okay to open a pull req.
```c
#define MPZ_NUM_DIG_FOR_INT (sizeof(mp_int_t) * 8 / MPZ_DIG_SIZE + 1)
#define MPZ_NUM_DIG_FOR_LL (sizeof(long long) * 8 / MPZ_DIG_SIZE + 1)
```

dpgeorge, 2015-09-15:
Fixed in 8b4fb4fe140e9cf57fcfa258d0d2d6fe19090fc5.

@alex-march I just beat you to it!  In this case the calculation was trying to round the division up, so the correct fix is (+ MPZ_DIG_SIZE-1) before the division.  Also I fixed the definition of sys.maxsize to make it more robust.

Yes, you are always welcome to open a PR :)

closed 2015-09-15 by dpgeorge

alex-march, 2015-09-15:
@dpgeorge good timing, thanks!