micropython/micropython #1450: Single precision floating point issues

dhylands, 2015-08-31:
While investigating #1435 I ran into some other formatting issues. The output isn't necessarily wrong, but it doesn't match CPython either (although I am concerned about the outputting more data than requested potentially causing a buffer overflow).

1 - Outputting more data than requested.
On pyboard:
```
>>> '{:.4e}'.format(9.99999e-5)
'1.00000e-04'
```
On CPython:
```
>>> '{:.4e}'.format(9.99999e-5)
'1.0000e-04'
```

2 - making the wrong choice about when to use e notation:
pyboard:
```
>>> '%5.g' % 9.99999
'   10'
```
CPython:
```
>>> '%5.g' % 9.99999
'1e+01'
```

3 - Failing to normalize when rounding:
pyboard:
```
>>> '%5.e' % 9.99999
'10e+00'
```
CPython:
```
>>> '%5.e' % 9.99999
'1e+01'
```

dpgeorge, 2017-06-14:
The first issue here has been fixed.  The other two relate to how rounding is done in mp_format_float: the rendering of the digits is done first and then rounding is applied later, and in the cases above the rounding actually changes how things should be rendered.  It looks like the only solution is to rewrite the function so that it does rounding before rendering the digits.

dhalbert, 2017-12-04:
Another example (found by a CircuitPython user):
```
>>> '%f' % 1111.0
'1110.999941'
>>> '%f' % 11111.0
'11110.999584'
>>> '%f' % 111111.0
'111110.997200'
>>> '%g` % 111111.0
111111
```
Notice how the accuracy gets progressively worse for `f` format. I believe the problem is that in `py/formatfloat.c`, the number is normalized to be between 1 and 10 by multiplying it by repeatedly by the values in `g_neg_pow[]`, which are values like 1e-1, 1e-2, 1e-4, etc., which don't have precise representations as floats. https://github.com/micropython/micropython/blob/155ec21e4933f75ee98bbd3e125e4304fb188170/py/formatfloat.c#L256

dpgeorge, 2017-12-04:
> I believe the problem is that in py/formatfloat.c, the number is normalized to be between 1 and 10 by multiplying it by repeatedly by the values in g_neg_pow[], which are values like 1e-1, 1e-2, 1e-4, etc., which don't have precise representations as floats.

Yes, I was investigating this recently and came to the same conclusion as what you write.  And it's not just single precision but also double.  Eg on the unix port one can see a threshold (ie change of behaviour in the formatter) going from 115 to 116 (note that an integer is used as the value to format, so it's nothing to do with float parsing, just formatting):
```
>>> '%f' % 115
'115.000000'
>>> '%f' % 116
'115.999999'
```

dpgeorge, 2018-03-01:
The cases of rounding when formatting values like 116, 1111, 11111 and 111111 are now fixed by bc12eca461a317df842ce2e616afa97670cd0ce3.  But note that for platforms that use single precision floats, `'%f' % 1111` will still render as `1110.999942` because without a precision specifier, `%f` will use 6 digits, and at 6 digits that's what a 32-bit 1111.0 looks like.  So one needs to use something like `%.3f` to get a nice round number.

yur-spiridonov, 2019-11-04:
The number 460.0000000000001 in the internal extended format will be represented by the binary code 111001100.00000000000000000000000000000000000000000001 | 111. If this number is rounded to the low bit, then we get 111001100.00000000000000000000000000000000000000000010 =
4.60000000000000113686837721616E2. If 16 significant decimal digits are displayed in a line, then we get 4.600000000000001.
If rounding is performed without Guard Digits, truncating all bits after the 53rd, then we will have 111001100.00000000000000000000000000000000000000000001 = 460.00000000000005684341886080801486968994140625. If we print 16 significant digits, we get the number 460.0000000000000. Maybe MicroPython and CPython can round binary numbers differently.

kevinjwalters, 2019-11-27:
We've just been chatting about `format()` peculiarities in CircuitPython land, from 5.0.0-alpha.5 on a new CPB board. An example:

```
>>> "{0:.20f}".format(9.00390625)  ### good
'9.00390625000000000000'
>>> "{0:.20f}".format(10.00390625)  ### odd
'10.00390648841857910156'
```

robert-hh, 2022-08-27:
Most of the examples mentioned here are now fixed.

dpgeorge, 2022-08-29:
The original issue (top post) listed 3 problems.  Problem 1 is fixed.  Problems 2 and 3 still have the same, wrong behaviour.

To close this issue, either Problems 2 and 3 are fixed, or we decide they don't need fixing.

projectgus, 2024-08-28:
Labelling as docs: the remaining issues seem to be both cosmetic and hard to fix reliably, so suggest the best solution is to document the difference to CPython.