micropython/micropython #1451: vm assert on "raise Exception from None" in unix builds

pramasoul, 2015-09-01:
Hits an assert in ```vm.c``` on unix builds:
```python
Micro Python v1.4.3-413-g21ffa7c on 2015-08-31; linux version
>>> raise Exception from None
micropython: ../py/vm.c:1051: mp_execute_bytecode: Assertion `unum <= 1' failed.
Aborted
```
```python
Micro Python v1.4.4-251-g21ffa7c on 2015-08-31; darwin version
>>> raise Exception from None
Assertion failed: (unum <= 1), function mp_execute_bytecode, file ../py/vm.c, line 1051.
Abort trap: 6
```
Works fine on pyboard:
```python
Micro Python v1.4.4-251-g21ffa7c on 2015-08-30; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> raise Exception from None
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: exceptions must derive from BaseException
```

pfalcon, 2015-09-01:
pyboard version is just built with asserts disabled, which causes it to run code for situations it doesn't expect, as your example shows (which isn't fine). We generally should get rid of asserts in the code and replace them with proper error reporting (which unfortunately will bloat code, so nobody hastes).

Regarding this specific case, we don't support exception chaining, but issuing a warning and ignoring "from" argument makes sense.

pfalcon, 2015-09-01:
Addressed in https://github.com/micropython/micropython/commit/2ff2ea5f3b1904834b287e74415f329b31d9a3db . The most boring part is fixing cascade of failing tests.

closed 2015-09-01 by pfalcon

pramasoul, 2015-09-01:
Naive of me to assume it worked fine on pyboard. Thanks for the explanation, and the fix.