micropython/micropython #1452: Error compiling with esp iot sdk version 1.3.0

wiliamsouza, 2015-09-04:
Got this error trying to compile micropython scroll to see the fix.

Error
------

```error
H -fno-inline-functions -Wl,-EL -mlongcalls -mtext-section-literals  -Os -DNDEBUG -fdata-sections -ffunction-sections -c -MD -o build/uart[149/452]
In file included from /home/wiliam/source/esp-open-sdk/sdk/include/ets_sys.h:12:0,
                 from uart.c:12:
uart.c: In function 'uart_config':
/home/wiliam/source/esp-open-sdk/sdk/include/eagle_soc.h:247:94: error: suggest parentheses around arithmetic in operand of '|' [-Werror=parenthese
s]
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)  do {WRITE_PERI_REG(PIN_NAME, READ_PERI_REG(PIN_NAME)&(~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S))|((((
FUNC&BIT2)<<2)|(FUNC&0x3))<<PERIPHS_IO_MUX_FUNC_S));}while(0)
                                                                                              ^
/home/wiliam/source/esp-open-sdk/sdk/include/eagle_soc.h:50:98: note: in definition of macro 'WRITE_PERI_REG'
 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)
                                                                                                  ^
uart.c:47:9: note: in expansion of macro 'PIN_FUNC_SELECT'
         PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO2_U, FUNC_U1TXD_BK);
         ^
/home/wiliam/source/esp-open-sdk/sdk/include/eagle_soc.h:247:94: error: suggest parentheses around arithmetic in operand of '|' [-Werror=parenthese
s]
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)  do {WRITE_PERI_REG(PIN_NAME, READ_PERI_REG(PIN_NAME)&(~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S))|((((
FUNC&BIT2)<<2)|(FUNC&0x3))<<PERIPHS_IO_MUX_FUNC_S));}while(0)
                                                                                              ^
/home/wiliam/source/esp-open-sdk/sdk/include/eagle_soc.h:50:98: note: in definition of macro 'WRITE_PERI_REG'
 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)
                                                                                                  ^
uart.c:51:9: note: in expansion of macro 'PIN_FUNC_SELECT'
         PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD);
         ^
/home/wiliam/source/esp-open-sdk/sdk/include/eagle_soc.h:247:94: error: suggest parentheses around arithmetic in operand of '|' [-Werror=parenthese
s]
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)  do {WRITE_PERI_REG(PIN_NAME, READ_PERI_REG(PIN_NAME)&(~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S))|((((
FUNC&BIT2)<<2)|(FUNC&0x3))<<PERIPHS_IO_MUX_FUNC_S));}while(0)
                                                                                              ^
/home/wiliam/source/esp-open-sdk/sdk/include/eagle_soc.h:50:98: note: in definition of macro 'WRITE_PERI_REG'
 #define WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val)
                                                                                                  ^
uart.c:52:9: note: in expansion of macro 'PIN_FUNC_SELECT'
         PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDO_U, FUNC_U0RTS);
         ^
cc1: all warnings being treated as errors
make: *** [build/uart.o] Error 1
```

patch
--------

```diff
--- esp_iot_sdk_v1.3.0/include/eagle_soc.h  2015-08-07 19:56:56.000000000 -0300                                                                    
+++ esp_iot_sdk_v1.3.0/include/eagle_soc.h  2015-09-03 21:30:14.823249805 -0300                                                                    
@@ -246,9 +246,9 @@                                                                                                                                
                                                                                                                                                   
 #define PIN_FUNC_SELECT(PIN_NAME, FUNC)  do { \                                                                                                   
     WRITE_PERI_REG(PIN_NAME,   \                                                                                                                  
-                                READ_PERI_REG(PIN_NAME) \                                                                                         
-                                     &  (~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S))  \                                                        
-                                     |( (((FUNC&BIT2)<<2)|(FUNC&0x3))<<PERIPHS_IO_MUX_FUNC_S) );  \                                               
+                                ((READ_PERI_REG(PIN_NAME) \                                                                                       
+                                     &  (~(PERIPHS_IO_MUX_FUNC<<PERIPHS_IO_MUX_FUNC_S)))  \                                                       
+                                     |( (((FUNC&BIT2)<<2)|(FUNC&0x3))<<PERIPHS_IO_MUX_FUNC_S)));  \                                               
     } while (0)                                                                                                                                   
                                                                                                                                                   
 //}}
```

wiliamsouza, 2015-09-04:
After deploy I getting this:

```
load 0x40100000, len 30680, room 16 
tail 8
chksum 0x07
load 0x3ffe8000, len 892, room 0 
tail 12
chksum 0x68
ho 0 tail 12 room 4
load 0x3ffe8380, len 18844, room 12 
tail 0
chksum 0x13
csum 0x13
����׮$.],H[Y[زX�XC�Fatal exception (0):
epc1=0x402352e0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
```

pfalcon, 2015-09-04:
Cannot reproduce.

wiliamsouza, 2015-09-04:
With ESP-01 using:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.3 LTS
Release:	14.04
Codename:	trusty
```

pfalcon, 2015-09-04:
But the main info is not that, but what Xtensa toolchain you use/where you got it from. If you follow port instructions, you're unlikely to hit issues (as least with building): https://github.com/micropython/micropython/blob/master/esp8266/README.md

wiliamsouza, 2015-09-04:
Follow intructions from https://github.com/pfalcon/esp-open-sdk and https://github.com/micropython/micropython/blob/master/esp8266/README.md

build sdk with `make STANDALONE=n`

```
$ xtensa-lx106-elf-gcc -v
Using built-in specs.
COLLECT_GCC=xtensa-lx106-elf-gcc
COLLECT_LTO_WRAPPER=/home/wiliam/source/esp-open-sdk/xtensa-lx106-elf/libexec/gcc/xtensa-lx106-elf/4.8.2/lto-wrapper
Target: xtensa-lx106-elf
Configured with: /home/wiliam/source/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/configure --build=x86_64-build_unknown-linux-gnu --host=x86_64-build_unknown-linux-gnu --target=xtensa-lx106-elf --prefix=/home/wiliam/source/esp-open-sdk/xtensa-lx106-elf --with-local-prefix=/home/wiliam/source/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --disable-libmudflap --with-sysroot=/home/wiliam/source/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG 1.20.0' --disable-__cxa_atexit --with-gmp=/home/wiliam/source/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpfr=/home/wiliam/source/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-mpc=/home/wiliam/source/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-isl=/home/wiliam/source/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-cloog=/home/wiliam/source/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --with-libelf=/home/wiliam/source/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools --enable-lto --enable-target-optspace --disable-libgomp --disable-libmudflap --disable-nls --disable-multilib --enable-languages=c,c++
Thread model: single
gcc version 4.8.2 (crosstool-NG 1.20.0) 
```

Trying build everything againg

pfalcon, 2015-09-04:
If you use STANDALONE=n, then you'll need to set ESP_SDK for uPy's Makefile. esp-open-sdk has default STANDALONE=y exactly because it's the most convenient option for most people.

wiliamsouza, 2015-09-04:
I changed this https://github.com/micropython/micropython/blob/master/esp8266/Makefile#L15 to point to my sdk folder. Is there other place to change?

wiliamsouza, 2015-09-04:
With `STANDALONE=y` it changed to `boot mode:(3,3)` sometimes.

```
load 0x40100000, len 30680, room 16 
tail 8
chksum 0xd8
load 0x3ffe8000, len 892, room 0 
tail 12
chksum 0xd3
ho 0 tail 12 room 4
load 0x3ffe8380, len 18832, room 12 
tail 4
chksum 0x0e
csum 0x0e
�����W$.�,�[Y[X�X_XC�Fatal exception (0):
epc1=0x402352e0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,3)
```

pfalcon, 2015-09-11:
I just built and booted latest micropython esp8266 master with latest esp-open-sdk (1.3.0). Everything works as expected.

pfalcon, 2015-09-11:
@mianos: Can you confirm which versions you use?

mianos, 2015-09-11:
Paul, I use the one pulled in from builder: https://github.com/pfalcon/esp-open-sdk.git
I rebuilt it  last week and I got esp_iot_sdk_v1.3.0 
Working fine for me, even with all my oddball C code that uses the SDK directly.

wiliamsouza, 2015-09-12:
I will try it again! wiring it like this https://github.com/Garagem-Hacker/senseiot/blob/master/ht/esp8266_esp_01_ftdi_logic_level_converter_dht11_bb.png

pfalcon, 2015-10-13:
Not reproducible - closing.

closed 2015-10-13 by pfalcon

drasko, 2016-03-05:
@pfalcon when you say "If you use STANDALONE=n, then you'll need to set ESP_SDK for uPy's Makefile."  what do you mean exactly? I.e. how should it be changed?

I am building SDK with STANDALONE=n, because I need this for some other projects, so how whould I change Micropython's Makefile?

drasko, 2016-03-05:
Actually found it, it should be something like `.../esp-open-sdk/esp_iot_sdk_v1.5.0`. Did not notice because of some other errors.