micropython/micropython #1454: Should 32-bit peripherals support mpz ints rather than just small ints?

dhylands, 2015-09-05:
This came up as a result of: http://forum.micropython.org/viewtopic.php?f=6&t=905

One way we could solve this is to have functions like timer.counter() generate mpz values when the answer won't fit in a small int. I think that it should also accept mpz values which are in range as well.

If we made a little helper function we could use this where ever numbers larger than a small int could be used. (i.e. stm.mem32 would be able to get/set the full 32-bit value).

It would be no worse for interrupt code than it currently is. To help out interrupt code (assuming we don't allow allocations) we could add counter_high and counter_low functions which retrieve the low/high portions of the 32-bit counter, or we could just say that you need to use stm.mem16 to get at those.

dbc, 2015-09-05:
I think returning an mpz when necessary is a good solution -- from the user's perspective, it passes the "Python works as expected with no surprises" test. I think accepting mpz values is important for peripheral-related constructors in order to utilize the underlying hardware's capabilities in the "obvious" manner.

I'm not deep enough into the code to understand the allocation-in-interrupts comment.  Would a handler written in Python not be able to call timer.counter() and get an mpz due to restrictions on allocations within interrupts? Or is this a restriction on the HAL-level C code?

dpgeorge, 2015-09-05:
There already exists the required functions to get/create integers, which uses mpz only if needed:
```
mp_obj_new_int(x) // will create small-int or mpz
mp_obj_new_int_from_uint(x) // will create small-int or mpz
mp_obj_get_int(o) // will raise if value does not fit in 32 bits signed
mp_obj_get_int_truncated(o) // will truncate mpz to 32 bits if needed
```
It's just a matter of using these in the right place.  At the moment the timer period fails because it's using mp_obj_get_int to extract the value (which doesn't fit in 32-bits signed).

dhylands, 2015-09-05:
> I'm not deep enough into the code to understand the allocation-in-interrupts comment. Would a handler written in Python not be able to call timer.counter() and get an mpz due to restrictions on allocations within interrupts? Or is this a restriction on the HAL-level C code?

Small ints (which can store up to 31 bits) don't require any memory allocations (since they fit in a register). qstr also don't require any memory allocations. All other objects require some type of allocation (which is why floats and mpz can't currently be used from inside interrupts).

You can preallocate memory for queues and the like, but even something simple like a = 1.5 requires allocation of a float object.

dhylands, 2015-09-05:
> There already exists the required functions to get/create integers, which uses mpz only if needed:
>
> mp_obj_new_int(x) // will create small-int or mpz
> mp_obj_new_int_from_uint(x) // will create [...]

So does this mean that perhaps we just need an mp_obj_get_uint (will raise if value does not fit into 32 bits unsigned) and perhaps the matching mp_obj_get_uint_truncated?

dbc, 2015-09-05:
@dhylands 
Yes, I knew all that.  My question is about the context in which the allocation restriction arises.  When you say "inside interrupts" does that mean "inside handlers"?

dhylands, 2015-09-05:
> Yes, I knew all that. My question is about the context in which the allocation restriction arises. When you say "inside interrupts" does that mean "inside handlers"?

Yes. Just before calling any python callback routine from within an interrupt handler, we "lock" the heap (which causes an exception to be raised if any attempt is made to allocate memory). This is done because the heap allocation/garbage collection code is not reentrant.

dbc, 2015-09-05:
It appears to me that pyb_timer_init_helper uses 0xffffffff as a NULL value to check for missing parameters.  The current code disallows a period of 0xffffffff due to over-eager parameter checking, even if you get past the mp_object_get... issue.

dhylands, 2015-09-05:
I think that if we supported mpz then we'd need to change the parameter from an int to a python object. Then None could be used as the missing parameter check.

dbc, 2015-09-05:
> Yes. Just before calling any python callback routine from within an interrupt handler, we "lock" the
> heap (which causes an exception to be raised if any attempt is made to allocate memory). This is 
> done because the heap allocation/garbage collection code is not reentrant.

Makes sense. Pre-allocating something to put the result in could work but is unPythonic and non-obvious. Reading a counter in two 16 bit halves has it's own issues because of the volatile nature of the register.

dbc, 2015-09-05:
While a bit of a hack, defining counter_low and counter_high functions with a nasty-hack side effect such that:

* counter_low() captures the 32 bit counter atomically and returns the low 16 bits
* counter_high() returns the high 16 bits of the most recently captured 32 bit counter

gives simple access to an atomic read of the 32 bit counter, and avoids having to allocate any mpz's in the handler.  It introduces an ugly order dependency in user code.  counter_low() should do the capture since if you only want half, that's the half you probably want.

dbc, 2015-09-07:
Certainly there must be other places where read access to a 32 bit value inside an interrupt-level callback function is an issue.  Is there a way to augment mem32[] to solve this problem generically? Perhaps there is a way to finesse returning a tuple?
```python
u16 = 0
l16 = 0
def someCallback():
    u16,l16 = mem32inHalves[someAddress]
```
The ultimate destination variables are allocated globally.  The tuple container I assume requires an allocation which can't be done at interrupt level.  Is it possible to pre-allocate a tuple framework for use within the "inHalves" version of the call, with a lifetime sufficient to return a tuple inside an interrupt handler?  A bit of a kludge, certainly, and I don't see the "inHalves" function being of use outside of callbacks.

dpgeorge, 2020-06-09:
I'll close this discussion as outdated.  Feel free to reopen with comments on how it relates to the current code.

closed 2020-06-09 by dpgeorge