micropython/micropython #1455: Unexpected Pyboard behaviour around pyb.stop() and rtc.wakeup()

peterhinch, 2015-09-07:
The following seems indicative of a bug. The script runs as expected only if a 10mS or longer delay is placed in the penultimate line. With a delay of 1mS the blue LED remains on almost continuously, only rarely going out for a second. Intermediate values result in erratic operation. The power consumption of the Pyboard indicates that the code is running every second yet the LED is not reliably extinguished. Puzzling as the LED is turned off before the function call.
```python
import pyb
rtc = pyb.RTC()
rtc.datetime((2015, 8, 6, 4, 13, 0, 0, 0))
usb_connected = pyb.Pin.board.USB_VBUS.value() == 1

def lpdelay(ms): # attempt at a low power delay
    rtc.wakeup(ms)
    pyb.stop()
    rtc.wakeup(None)

if not usb_connected:
    led = pyb.LED(4)
    while True:
        led.on()
        pyb.delay(1000)
        led.off()
        pyb.delay(1) # why?
        lpdelay(1000)
```

dpgeorge, 2015-09-07:
LED(4) is PWM so the timer might not be getting a chance to clock over and turn the PWM off before you go to sleep (pyb.stop() will stop the timer counting).  Try LED 1,2,3.

dhylands, 2015-09-07:
Makes sense now. You could really make it go off by configuring the pin as a gpio with a value of zero.

For any intensity value other than 0%, you'd still get indeterminate behaviour, and as soon as the pyb.stop was issued (assuming you didn't turn the LED off) the LED would either go 0% or 100% depending on where in the PWM cycle things were.

peterhinch, 2015-09-08:
I can confirm it runs correctly with no delay with the other LED's. Thanks for explaining this puzzling (to me) behaviour occasioned by my entirely arbitrary choice of test LED.

peterhinch, 2015-09-09:
Perhaps my example is too far off-piste to warrant code changes, but an option in the new API might be to instantiate LED's with a USE_PWM argument defaulting False.

dpgeorge, 2015-09-09:
A more transparent change/fix would be to only use PWM if the intensity method is called.  And maybe .on() and .off() can go back to normal GPIO to do a proper on/off.

dpgeorge, 2016-01-29:
This is now fixed by a8a4b01af6501ff7ab61b251fc30fcf8a286f8d1.  PWM for LEDs is now only activated if led.intensity() is called with a number between 1 and 254 inclusive.  led.on() and led.off() will always configure the pin as a GPIO and turn it on/off that way.

closed 2016-01-29 by dpgeorge