micropython/micropython #1461: CC3100 Driver

thinkl33t, 2015-09-15:
Has any work been done on a CC3100 driver for the stmhal core?  This would provide an upgrade path from the (frankly terrible) CC3000 chip, and I understand the majority of the work has already been done for the cc3200 core (iirc the CC3200 is an arm with cc3100 onboard).

Devboards are available (CC3100MODBOOST is a good option, as its only £15), as are plain chips and modules that can be easily integrated (CC3100MOD is available from mouser, and can be sampled directly from TI)

I found https://github.com/blmorris/micropython-cc3100 when I was looking to see if this had been done, but its licensing is unclear, and it cant be merged cleanly with modern micropython (i think the wlan API has changed since this was written?)

blmorris, 2015-09-15:
Hi there! So that is my repo, but to be clear, very little of it is my work.

The short answer to your question is that some work has been done on a CC3100 driver, but all of the relevant stuff has been integrated into the CC3200 port by @danicampora, who has taken it far beyond what is in my repo. It is my understanding that the plan is to eventually factor out a CC3100 driver for stmhal from the CC3200 code, but for now the priority has been to develop the CC3200 port and get the network API to a more consistent and mature state.

The Longer Answer: Sometime last October or November I inquired on the MicroPython forum if anyone had done any work on the CC3100. I got a response from Nadim El-Fata, who had gotten it to work for his own purposes but was moving on to other projects. He was happy to share it with me, with the only request being that I post it on GitHub for him, which I was more than happy to do. (I never thought about appropriate licensing). A few months later he sent me some updated code, and again with his permission I integrated it into a build-able uPy source tree and pushed it to GitHub; the resulting branch is here: https://github.com/blmorris/micropython/tree/cc3100-dev

I built this branch specifically to share with the core developers, and to the best of my knowledge pretty much anything that would be usable has been integrated into the CC3200 / WiPy port. @dpgeorge at one point had a branch which provided a CC3100 driver for stmhal, but I don't think this was ever cleaned up for merging; the priority has been to develop the WiPy. Given the progress that has been made in the CC3200 / WiPy port, the best way to proceed now would probably be to make a new CC3100 driver for stmhal based on the CC3200 code. I have no plans to work on that at this point, so I would welcome anyone who is interested to give it a shot.

-Bryan

dpgeorge, 2015-09-15:
Yes, @blmorris has it correct.  I almost got the CC3100 working but there were some (deep?) issues with the interrupts and calling the CC3100 periodically to keep it running.  So it was never merged.  It needs someone to dedicate a bit of time to get working.

dpgeorge, 2020-06-09:
Now that other wifi drivers are available for stm32, mainly CYW4343, it's very unlikely that the CC3100 will be supported.  So I'll close this issue.

closed 2020-06-09 by dpgeorge