micropython/micropython #1465: mips64 build error

cpackham, 2015-09-18:
Hi,

After seeing micropython showed off at kiwipycon I thought I'd get the unix port integrated into [buildroot](http://buildroot.org/). As of yesterday it's [available](http://lists.busybox.net/pipermail/buildroot/2015-September/139804.html) as part of buildroot.

Unfortunately this has come with a few build errors for some architectures [xtensa](http://autobuild.buildroot.org/results/670/670907760d6b3e7eb46b433647b392cec3bafdcd/build-end.log), [sh4a](http://autobuild.buildroot.org/results/f5a/f5a2c563b8bdab9e345960a5b92d31f3f92dba00/build-end.log) and [mips64](http://autobuild.buildroot.org/results/99a/99a610c72877fef53d93e1a923bd5c71b3376618/build-end.log).

The first 2 I think are just a matter of passing -DMICROPY_GCREGS_SETJMP=1 in the build, I'm working on a buildroot patch for that now. The last one is probably more to do with being compiled for a 64 bit processor, I haven't had a good look yet but I thought I'd raise this here in case someone else was able to take a look.

Thanks.

cpackham, 2015-09-18:
Looks like defining -DMPZ_DIG_SIZE=32 when the target is 64 bit does the trick

dpgeorge, 2015-09-18:
The issue on mips64 might be fixed by 8b4fb4fe140e9cf57fcfa258d0d2d6fe19090fc5.

But really it would be good to know the compiler macro definition that allows to detect 64-bit MIPS, so we can update py/mpz.h to automatically use MPZ_DIG_SIZE=32 for this arch.  Do you know it?

cpackham, 2015-09-19:
Can't say that I know it and just to complicate things someone on the buildroot list also pointed out that there is such a thing as 64-bit MIPS CPU with a 32-bit ABI. For what it's worth the compiler is gcc but I can't find a definitive reference for the marco definitions used.

I'll give 8b4fb4f a go to see if that fixes the problem.

pfalcon, 2015-09-19:
@cpackham : Great news, thanks!

To figure out predefined macros for a particular toolchain, see http://stackoverflow.com/questions/2224334/gcc-dump-preprocessor-defines . Just quote toolchain name and output here and we hopefully will figure it out.

@dpgeorge : On the other hand, why 16-bit MPZ_DIG doesn't work with 64-bit archs? Maybe that's what needs fixing?

cpackham, 2015-09-19:
@pfalcon all the gory details of all the buildroot toolchains are available [here](http://autobuild.buildroot.org/toolchains/). For the mips64 compiler I've put the output [here](https://gist.github.com/cpackham/653edb4fddba6205c863) there are a few things that look useful but there still might be EABI differences that need to be accounted for.

dpgeorge, 2015-09-19:
> On the other hand, why 16-bit MPZ_DIG doesn't work with 64-bit archs? Maybe that's what needs fixing?

No, that should work (any bit size of 4 or more should work on any arch).  It was just a warning/error with sys.maxsize constant, which should now be fixed.

cpackham, 2015-09-20:
Yep 8b4fb4f does the trick. I'll work to add that as a patch on top of v1.4.5 in buildroot.

closed 2015-09-20 by cpackham