micropython/micropython #1466: nrf24l01 driver SPI problem

peterhinch, 2015-09-18:
This driver initialises the SPI bus with polarity=0, phase=1. This produces the following waveform (red SCK, yellow MOSI)
[image]
This contradicts the datasheet which shows MOSI changing on the clock negative edge, and being read on the positive edge.
[image]
Setting phase=0 produces the correct waveform, and results in more reliable SPI communication.
[image]

dpgeorge, 2015-09-18:
Yes I think you're right, it should be phase=0.

This error is unfortunately a recurring theme: a while back 480a7ce58f6e5dc6179373f1f8b69d6a5434b2d4 the phase convention was changed from 1/2 to 0/1 to match industry standard (see #936).  The existing drivers (like nrf24) were not updated with the change.

dpgeorge, 2015-09-18:
Fixed in b0c08c8c173cbfe66ad78af55f4025543bbdbd9b.

closed 2015-09-18 by dpgeorge