micropython/micropython #1475: Prepare to 1.4.6 release?

pfalcon, 2015-09-21:
There're changes like #1470 which noticeably change build process. Should we tag 1.4.6 (>170 commits since 1.4.5), to let more time for changes like above to be testes, stabilize, documented, etc.?

dpgeorge, 2015-09-22:
Yes I agree go to 1.4.6 before #1470 is merged.

What about the new hardware API #1430... and shipping of WiPy?  @danicampora how about we tag 1.4.6 before shipping, is that possible or not anymore?

danicampora, 2015-09-22:
@dpgeorge yes, it is possible, please go ahead. We still have a few days work before shipping can start.

dpgeorge, 2015-09-22:
Ok.  Please list anything that wants in before 1.4.6 and then it'll be tagged within the next 24 hours.

danicampora, 2015-09-22:
Nothing from my side.

pfalcon, 2015-09-23:
Nothing specific from me too.

dpgeorge, 2015-09-23:
Done in 5572f735b65cb4704725966d5437564501ceb673.

closed 2015-09-23 by dpgeorge