micropython/micropython #1476: Error when calling func(*args, kw_arg=None)

dbrignoli, 2015-09-22:
Compiling the unix port from the current master branch, the snippet:
```
def func(*args, **kwargs):
    pass

args = ['a','b']
func(*args, kw_arg=None)
```
throws:
```
Traceback (most recent call last):
  File "../tests/basics/starargs.py", line 6, in <module>
TypeError: 'NoneType' object is not iterable
```

Looking at the bytecode below and the comment [here](https://github.com/micropython/micropython/blob/1b693543aa5dcc3c5d35c63f4d657689d2f8fc1e/py/vm.c#L912) I am thinking that maybe arguments are being passed in the wrong order...

```
> ./micropython -v -v ../tests/basics/starargs.py
File ../tests/basics/starargs.py, code block '<module>' (descriptor: 0xb73ef020, bytecode @0xb73ef145 51 bytes)
Raw bytecode (code_info_size=12, bytecode_size=39):
 0c 82 2e 83 61 23 68 2a 00 00 00 00 05 00 ff 60
 00 f1 3e b7 24 83 31 16 83 63 16 83 64 51 02 24
 69 1c 83 31 00 1c 69 00 16 83 65 11 18 65 82 00
 32 11 5b
arg names:
(N_STATE 5)
(N_EXC_STACK 0)
  bc=-3 line=1
  bc=0 line=2
  bc=8 line=5
  bc=18 line=6
00 MAKE_FUNCTION 0xb73ef100
05 STORE_NAME func
08 LOAD_CONST_STRING 'a'
11 LOAD_CONST_STRING 'b'
14 BUILD_LIST 2
16 STORE_NAME args
18 LOAD_NAME func (cache=0)
22 LOAD_NAME args (cache=0)
25 LOAD_CONST_STRING 'kw_arg'
28 LOAD_CONST_NONE
29 LOAD_NULL
30 CALL_FUNCTION_VAR_KW n=0 nkw=1
33 POP_TOP
34 LOAD_CONST_NONE
35 RETURN_VALUE
File ../tests/basics/starargs.py, code block 'func' (descriptor: 0xb73ef100, bytecode @0xb73ef185 13 bytes)
Raw bytecode (code_info_size=8, bytecode_size=5):
 08 83 31 83 61 43 00 00 03 00 ff 11 5b
arg names:
(N_STATE 3)
(N_EXC_STACK 0)
  bc=-3 line=1
  bc=0 line=3
00 LOAD_CONST_NONE
01 RETURN_VALUE
Traceback (most recent call last):
  File "../tests/basics/starargs.py", line 6, in <module>
TypeError: 'NoneType' object is not iterable
mem: total=2092, current=724, peak=1550
stack: 0 out of 40000
GC: total: 1024512, used: 800, free: 1023712
 No. of 1-blocks: 14, 2-blocks: 2, max blk sz: 8
```

dpgeorge, 2015-09-22:
This looks like a bug in the compiler (!).  I don't think we have a test for this case (keyword arg following *args).

danicampora, 2015-09-22:
Interesting!

dbrignoli, 2015-09-22:
@dpgeorge, can you point me to the general area where function call argument lists are compiled? I tried (very briefly) to find it myself but I couldn't.

dpgeorge, 2015-09-22:
@dbrignoli I'd start here: https://github.com/micropython/micropython/blob/master/py/compile.c#L2176

dbrignoli, 2015-09-22:
Thanks @dpgeorge, my alter ego just opened #1477 for you to review.

dpgeorge, 2015-09-23:
Fixed by #1477.

closed 2015-09-23 by dpgeorge