micropython/micropython #1482: SD card driver

peterhinch, 2015-09-29:
The init_card() method currently makes two attempts to initialise the card before raising an OSError. I am finding that, with Sandisk Ultra 16GB cards, three attempts are required after an initial power up. I am currently using the following which so far has always initialised without error.
```python
        # CMD0: init card; should return R1_IDLE_STATE (allow 5 attempts)
        for count in range(5):
            if self.cmd(0, 0, 0x95) == R1_IDLE_STATE:
                break
        else:
            raise OSError("no SD card")
```
I have no idea why you have to keep hitting the thing :)

dpgeorge, 2015-09-29:
Does clocking the bus initially with more cycles help?  Ie the lines:
```python
        # clock card at least 100 cycles with cs high
        for i in range(16):
            self.spi.send(0xff)
```
if you make 16 a bigger number?

peterhinch, 2015-09-30:
Increasing it to 64 made no difference: it still takes 3 attempts.

dpgeorge, 2015-10-09:
Increased to 5 retries in 24652228af186d257bd273ddc8a4e07689d1a987.

closed 2015-10-09 by dpgeorge