micropython/micropython #1488: Possible errors in rtc.c

peterhinch, 2015-10-06:
The function pyb_rtc_wakeup() appears to contain errors in its handling of long delays. Because of the lengthy times involved, submitting a test case would be unhelpful but in my view this code should be reviewed. Note that the existing comments at the start of the function do conform with the chip data. Here is my idea of how the code should look - comments give reasons.
```
            if (div <= 16) { # Existing code is correct until the if statement
                wut = 32768 / div * ms / 1000;
            } else {
                wucksel = 4;
                wut = ms / 1000; # clock is now 1Hz
                if (wut >= 0x10000) { # doesn't fit in 16 bit register
                    wucksel = 6; # see comments at start of this function: adds 0x10000
                    wut -= 0x10000; # Subtract the 0x10000 added to wut by wucksel
                    if (wut >= 0x10000) { # still won't fit
                        nlr_raise(mp_obj_new_exception_msg(&mp_type_ValueError, "wakeup value too large"));
                    }
                }
            }
            if wut > 0: # evil case where ms/1000 == 0x10000
                {
                wut -=1
                }
```

dhylands, 2015-10-06:
Having tests is still reasonable, but I wouldn't necessarily test the amount of time of the actual sleep. Instead, I would test the register values programmed for a given sleep period.

So either, call pyb.wakeup and then use stm to read back the registers to verify that they were programmed correctly, or write a special test function which just does the register calculation and returns the results. Both the test function and wakeup would call the same function.

Doing the register thing is probably preferred since it means you don't have to expose a test function which is otherwise not that useful.

dpgeorge, 2015-10-06:
I agree that it should be `wucksel=6` for the last part.  But you don't need to use `>=0x10000` because a desired value of `wut=0x10000` translates to a register value of `0xffff`.

peterhinch, 2015-10-07:
Good point, so you're suggesting
```C
                if (wut > 0x10000) { // won't fit in 16 bit register
                    wucksel = 6; // see comments at start of this function: adds 0x10000
                    wut -= 0x10000; // Subtract the 0x10000 added to wut by wucksel
                    if (wut > 0x10000) { // still won't fit
                        nlr_raise(mp_obj_new_exception_msg(&mp_type_ValueError, "wakeup value too large"));
                    }
                }
            }
            if (wut > 0){ // evil case where ms/1000 == 0x10000
                wut -=1;
                }
```
Apologies for lapsing into Python above.

dpgeorge, 2015-10-07:
That looks about right. Except that the final check for `wut>0` is only needed for the `div<=16` path, not the 0x10000 path.

peterhinch, 2015-10-07:
Are you sure? Unless I'm missing something there's a pathological case in the 0x10000 path.
If ms == 0x10000 *1000 then wut == 0x10000. When wut -= 0x10000; is executed you're left with a value of zero which needs testing.

dpgeorge, 2015-10-07:
If wut == 0x10000 then the subtraction is not executed with a > comparison.

peterhinch, 2015-10-07:
Doh! Brain fade on my part...

dpgeorge, 2015-10-08:
Now fixed.  Thanks @peterhinch!

dpgeorge, 2015-10-08:
PRs for tests to check the WUT register value use stm module, as suggested by @dhylands, are very welcome!

closed 2015-10-08 by dpgeorge

peterhinch, 2015-10-08:
I've used the following script to check the registers.
```python
def ms_set(): # Decodes outcome of setting rtc.wakeup().
    dividers = (16, 8, 4, 2)
    wucksel = stm.mem32[stm.RTC + stm.RTC_CR] & 7
    div = dividers[wucksel & 3]
    wut = stm.mem32[stm.RTC + stm.RTC_WUTR] & 0xffff
    clock_period = div/32768 if wucksel < 4 else 1.0  # seconds
    period = clock_period * wut if wucksel < 6 else clock_period * (wut + 0x10000)
    return 1000 * period
```
The result doesn't always always exactly match the value passed to wakeup(), notably with the 1Hz clock, because of the limitations of precision of the registers but with values < 5 minutes it produced convincing results with yesterday's build. I'll run it against your new build tomorrow.

peterhinch, 2015-10-09:
Results look good. I'll update the rtc.py test script using this (@dhylands) approach and submit a PR.