micropython/micropython #1489: Documentation of pyb.standby() appears incorrect

peterhinch, 2015-10-06:
The docs state:
"pyb.standby()
Put the pyboard into a “deep sleep” state.
This reduces power consumption to less than 50 uA. To wake from this sleep state requires an external interrupt or a real-time-clock event. Upon waking the system undergoes a hard reset."

The reference manual (RM0090) Table 23 (Section 5.3 - Low Power Modes on page 126) seems to indicate that EXTI can only wake the processor up when it's in stop mode, and not when its in standby mode.

My reading of the manual indicates that the only external interrupt capable of being supported by the Pyboard is a tamper interrupt on pin X18.

dpgeorge, 2015-10-06:
I agree.  Can only be woken by 2 pins: WKUP (PA0=X1) or TAMP1 (PC13=X18).

dpgeorge, 2015-10-09:
Fixed in 845b5a2a582023caf484f87e6f12abc715f6b922.

closed 2015-10-09 by dpgeorge