micropython/micropython #1491: pyboard.py fails to enter raw_repl if boot.py prints anything

dhylands, 2015-10-08:
If my boot.py looks like:
```
import pyb
uart = pyb.UART(6,115200)
pyb.repl_uart(uart)
print("REPL is also on UART 6 (Y1=Tx Y2=Rx)")
```

then when I do a Control-D I see:
```
PYB: sync filesystems
PYB: soft reboot
REPL is also on UART 6 (Y1=Tx Y2=Rx)
Micro Python v1.4.5-143-gdcbe936-dirty on 2015-09-22; PYBv1.0 with STM32F405RG
Type "help()" for more information.
```

However, when I then try to use pyboard.py to run a simple hello.py program which just does:
```
print('Hello World')
```
then I get the following error:
```
518 >./pyboard.py hello.py 
OK
PYB: sync filesystems
PYB: soft reboot
REPL is also on UART 6 (Y1=Tx Y2=Rx)
raw REPL; CTRL-B to exit
>
could not enter raw repl
```

It seems that pyboard.py ONLY works if boot.py doesn't print anything at all, and main.py exits to the REPL.

It occurred to me, that we might not boot.py to run at all when entering the raw-repl or that we might want boot.py to run in a different mode.

For example, when I used the boot-switch.py boot.py file (https://github.com/dhylands/upy-examples/blob/master/boot_switch.py) with rshell, then it would go through the whole complicated flash the leds and have to wait for the timeout multiple times for each command executed.

So then I thought maybe it would be nice to have some way for boot.py to know that it was being run to enter a raw-repl versus a regular repl. The way the code is currently structured this isn't really possible. The one way I could think of doing it would be to make a Control-D variant which did a soft-reboot, but with an implied enter raw repl. Then we could set something for boot.py to use, and it could choose to do a simplified initialization and perhaps not bother executing main.py etc.

So I thought I put this up for discussion to see what opinions others had.

dhylands, 2015-10-08:
Looking at the code again, it looks like pyboard.py does the following:
```
Control-C Control-C # kill main
Control-A # Enter RAW REPL
Control-D # Do a soft reset (stays in raw repl)
```
So it may be as simple as exposing pyexec_mode_kind so that boot.py can read it.

dpgeorge, 2015-10-20:
Yes, this could be fixed/improved.  The aim with boot.py is that it's supposed to be minimal, but I can see that sometimes it needs to have complex logic in it.

dpgeorge, 2015-12-06:
This is now fixed by #1680.  Is it still useful to expose pyexec_mode_kind?

dhylands, 2015-12-06:
> This is now fixed by #1680. Is it still useful to expose pyexec_mode_kind?

I'd be happy to say that I don't think we need it anymore.

Since we'll eventually support machine.reset_cause, I think that having RAW_REPL as one of the potential reset causes would cover this is a more generic manner.

So I'll go ahead and close #1492 with a note about that.

dpgeorge, 2021-04-30:
This was fixed a while ago, see above.  So I'll close the issue.

closed 2021-04-30 by dpgeorge