micropython/micropython #1498: Doc still shows "import pyb" instead of "import machine"

Bittencourt, 2015-10-10:
Hi Guys,

I`m new here, so sorry if this issue is already being addressed or I`m simply looking at the wrong site for the docs of the cc3200 version (http://micropython.org/resources/docs/en/latest/wipy/index.html).
but the "pyb" lib was changed in the code to "machine" but not in docs.

danicampora, 2015-10-10:
You are completely right, docs are outdated, we'll update them once done with shipping!

Bittencourt, 2015-10-10:
Thank you for the quick answer.

Once I become more used to the code I will be glad to help on those minor fixes.

danicampora, 2015-10-10:
That would be awesome! Thanks :-).

pfalcon, 2015-10-25:
@danicampora : I believe I read changelog that this was fixed?

danicampora, 2015-10-25:
Yes, it is fixed now, thanks.

dpgeorge, 2015-10-25:
Docs have been re-generated from source so looks like it's now fixed.

closed 2015-10-25 by dpgeorge