micropython/micropython #1499: REPL don't start via UART on CC3200 version

Bittencourt, 2015-10-10:
Hi,

I had a small issue using the REPL terminal via UART on version 1.0. After successfully flashing (verified by accessing the REPL via telnet) the uart 0 is mute. By the documentation i considered that the terminal would be accessible by uart0 on boot.

I'm using a CC3200 launchpad board. If there is config I should enable for that may be a good idea to put on the readme file on the cc3200 folder of the project.

Thank you,

danicampora, 2015-10-10:
With the CC3200 there's only one official REPL, which is the telnet one. If you one terminal duplication on the UART do:

```python
from machine import UART
import os
uart = UART(0, 115200)
os.dupterm(uart)
```

But yes, you are right, this should be documented, any help is welcome ;-)

Cheers,
Daniel

Bittencourt, 2015-10-11:
Thank you, worked fine. 

I will try to add this to readme file, but I think it will be necessary a pull request.
And I'll be glad to help.

closed 2015-10-11 by Bittencourt

danicampora, 2015-10-11:
Yes, a pull request is the way to go, thanks.